google/openhtf #14: SIGINT (sent with CTRL+c) doesn't always stop the test.

jettisonjoe, 2015-09-01:
Usually it does, but not always. I haven't identified the specific conditions that cause SIGINT handling to break yet.

fahhem, 2015-09-01:
This is usually a threading problem, they don't always like to die on their own. You can either set them to daemon mode or explicitly pass some "die now" data into them

jettisonjoe, 2015-09-01:
Yeah, I suspect the threads that aren't dying are in the Rocket instance in the HttpHandler. Once we flatten multi-cell support in the framework, there will be fewer places to look for the cause of this behavior.

jettisonjoe, 2015-10-21:
More info on this. It actually still happens with the bare-bones frontend API server we now have, so it wasn't specific to Rocket. SIGINT handling consistently fails if SIGINT happens while execution is inside one of the HTTP handlers.

wallacbe, 2015-10-24:
Provide a DUT ID in order to start the test.
^CReceived SIGINT. Stopping everything.
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 57906)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 647, in __init__
    self.setup()
  File "/usr/lib/python2.7/SocketServer.py", line 699, in setup
    self.rfile = self.connection.makefile('rb', self.rbufsize)
  File "/usr/lib/python2.7/socket.py", line 217, in makefile
    return _fileobject(self._sock, mode, bufsize)
ThreadTerminationError

wallacbe, 2016-01-22:
Observing this also with the bazel build environment

wallacbe, 2016-02-29:
Found that  SIGQUIT  ( Ctrl+\) will kill the process.

fahhem, 2016-03-19:
Does this only happen when the frontend is being accessed or can it be any time?

wallacbe, 2016-03-19:
My test cases about were without the front end running.

Also SIGQUIT may not be best method since tear down methods would be skipped.

wallacbe, 2016-05-17:
Additional observation from head:

If I start the tester like this then do CTRL+C it hangs
test.Execute(test_start=triggers.PromptForTestStart())

If I disable http and execute CTRL+C stops execution immediately:
test.Configure(http_port=None)
test.Execute(test_start=triggers.PromptForTestStart())

Does this point to http server being the issue?

jettisonjoe, 2016-05-19:
Yes, I believe the issue is in the http server. @grybmadsci and I have some changes planned for the HTTP API that will give us a good opportunity to investigate and resolve this.

jettisonjoe, 2016-10-10:
Is anyone still seeing issues with `CTRL+c` in the current PyPI version?

jettisonjoe, 2018-05-05:
I believe this is fixed now.

closed 2018-05-05 by jettisonjoe