google/openhtf #15: Add unit tests!

jettisonjoe, 2015-09-01:


wallacbe, 2015-11-11:
Also include unit test example for phases in hello_world tester

wallacbe, 2015-12-08:
I'm trying to implement a unit test on a phase.  I'm able to mock the decorators but unsure how to mock   config = conf.Config() with it being global

fahhem, 2015-12-08:
Just mock it out and then end the mocking when you're done. There are a few ways to do this, we can discuss over chat if you wish

wallacbe, 2016-03-11:
Looking for unit test example of how we test for a FAIL case of a phase where a measurement is out of Range().  Would we be able to this on a phase level or would we have to do it on a tester level?

glados-verma, 2022-10-12:
Closing this because some unit tests were added :joy: More seriously though, we can always improve unit test coverage, and intend to track this with coverage tools.

closed 2022-10-12 by glados-verma