google/openhtf #4: Log a warning on setting a measurement to 'None'.

jettisonjoe, 2015-09-01:
This might indicate an 'ERROR' outcome in some scenarios. For instance, if we set a measurement using the output of some hardware plug. Logging a warning in this scenario gives us a breadcrumb in case we don't validate the measurement in this way.

grybmadsci, 2016-04-04:
Hm, I would contend that we should disallow this entirely.  None is a python-specific thing, and while the framework is in Python, I think we've tried pretty hard to make the test result record language agnostic.  I think setting a measurement to None should raise.

fahhem, 2016-04-04:
I would consider setting to None the same as being cleared, so it would be an ERROR in the end in the current scheme. How about that?

jettisonjoe, 2016-04-05:
I think even if we support 'clearing' of measurements with `None`, we still probably want to at _least_ log a warning when doing so.

wallacbe, 2016-06-05:
Please close

closed 2016-06-05 by grybmadsci