google/openhtf #38: Maybe clear lint error having to do with 'default' in OutputToJSON.

jettisonjoe, 2015-10-19:
openhtf/__init__.py line 71

Yields the following lint error:
`An attribute defined in json.encoder line 162 hides this method.`

fahhem, 2016-03-23:
What is the pylint regimen? It's not documented in CONTRIBUTING.md and running "pylint openhtf" generates an overwhelming number of errors despite the generous sprinkling of pylint disabling comments in the code.

grybmadsci, 2016-03-23:
lol, "generous sprinkling" is a good way of putting it... i think the current state is the result of a "best effort" policy wrt pylint.  I'd be up for a pylint-sprint to clean up all those errors, but I'd like to first do an audit of the pylintrc file and make sure the result is what we really want.

I'll leave that to Joe in a separate issue.  Going to close this issue; it's woefully out of date now, and as Farz says, masked by the otherwise overwhelming number of lint errors.

closed 2016-03-23 by grybmadsci