google/openhtf #44: Write user docs for v1.0.

jettisonjoe, 2015-10-21:
We have this nice wiki feature. We should actually consider using it.

jettisonjoe, 2016-07-08:
We should look into https://readthedocs.org/ for this.

DanLipsitt, 2017-08-09:
> We'd be happy if you start a docs/ folder :)
– @fahhem in #613

If I create a config compatible with readthedocs, will a project member [create an account](https://readthedocs.org/accounts/signup/) there and [link it](https://docs.readthedocs.io/en/latest/getting_started.html#sign-up-and-connect-an-external-account) to this repo so the docs get built and hosted?

fahhem, 2017-08-13:
Yup, 100% I will! I already have 2 of my personal repos on there

fahhem, 2017-08-13:
Oh, looks like @maddychan already did that. Any chance she can add Joe and myself as owners? https://readthedocs.org/projects/openhtf/

DanLipsitt, 2017-08-21:
I've got autogenerated doc builds working in #641. I'm using [PBR](https://docs.openstack.org/pbr/latest/) and right now it seems to be interfering with protobuf builds. I'm pretty sure the solution is to move some of the config from `setup.py` to `setup.cfg` so that PBR can access it. Before I invest more in PBR, I want to see how you feel about using it. I think it has several benefits in addition to making Sphinx easier. The alternative to using it seems to be to add configuration to run `sphinx-autogen` automatically or to remember to run it manually whenever code changes.

wizofe, 2018-02-27:
I am considering on using OpenHTF to test our Raspberry Pi3 based Linux "flavour"/distro and could be more than happy to offer some help with the documentation or anything around. Let me know.