google/openhtf #46: logging.info not showing up in console output

wallacbe, 2015-10-24:
Not seeing logging.info() show up in the console output.

In the example I don't see the "Tearing Down" or if I add logging.info to the phase I don't see output.

amyxchen, 2015-10-26:
Console output default level is warning. if you want to see the info level logging, use flag "--verbosity info".  

--verbosity is for console level logging, for log file level, use "--log_file" to set the log file name, use "--log_level"  to set the log file logging level. (debug, info, warning, error, critical).

wallacbe, 2015-11-06:
I added --verbosity info and I see the INFO about starting the phase
However I don't see the output from within the phase test.logger.info('blah')

jettisonjoe, 2015-11-12:
We separated framework versus tester logging by design, and the command line flags were set up to control framework logging, with tester logging going to the test record (and from there to the output module). Maybe we can talk about this at our next in person meeting and see how we should handle the need for tester console output.

wallacbe, 2016-01-28:
I've been adding the following to my phases and I'm able to see the output.  Maybe openhtf can do something similar?

import logging
logging.basicConfig()
logger = logging.getLogger()
logger.setLevel(logging.INFO)

wallacbe, 2016-05-23:
Ran from head, I see output in the expected places, closing issue

closed 2016-05-23 by wallacbe