google/openhtf #49: Front end shows Internal Server Error when launching Front End from wheel installation

wallacbe, 2015-11-06:
I'm trying to create a wheel for OpenHtf to make deployment easier.

When I clone OpenHTF and pip install and start the front end I see it come up on localhost:5000.
When I create a wheel and pip install and start the front end I see it Running, but the webpage comes up as Internal Server Error.

jettisonjoe, 2015-11-12:
Hmm. Wonder why. I'll investigate.

jettisonjoe, 2015-12-09:
The frontend changes I'm working on will replace Flask's dev server with tornado's (proper) web server. This issue will likely evaporate then.

wallacbe, 2016-01-27:
I tested PR Multicast station discovery service. #59.

I'm able to launch frontend with a pre-built python wheel within bazel build environment.  Marking as closed.

closed 2016-01-27 by wallacbe