google/openhtf #60: Add ability to launch without config file

wallacbe, 2015-11-30:
Add ability to launch without config file.  Add ability to pass a required configs through command line.

  --config: The OpenHTF configuration file for this tester
    (default: '/usr/local/openhtf_client/config/clientfoo.yaml')
  --config_value: Allows specifying a configuration key=value on the command line. The format should be --config_value key=value. This value
    will override any existing config value at config load time and will be a string;
    repeat this option to specify a list of values
    (default: '[]')

jettisonjoe, 2015-12-09:
It should also be the case that if someone just runs ```~$ python test_script.py``` (no config, no flags) something reasonable should happen, not just an error.

fahhem, 2015-12-09:
I believe --config_yaml was discussed as a better alternative to --config_value.

jettisonjoe, 2015-12-09:
True, this issue's description runs at odds to what was discussed in the other thread, but the spirit of it is still something we should address, namely it should be possible to kick off a test script with no flags and no config and have something other than an error happen.

wallacbe, 2016-01-22:
Notes from PR:
Add default option to conf Declare function
provide a default station ID

wallacbe, 2016-03-10:
Current behavior:

1. station_id is not required in config file

2. config file must exist otherwise: openhtf.conf.ConfigurationMissingError:

3. config must be populated with at least one dummy value like 'foo: bar' otherwise: openhtf.conf.ConfigurationInvalidError: ('No data'...

grybmadsci, 2016-03-10:
You can workaround this by calling conf.LoadFromDict({}), if that helps in the immediate future, @wallacbe but we should definitely fix this properly.

wallacbe, 2016-05-23:
This is supported, closing issue

closed 2016-05-23 by wallacbe