google/openhtf #67: config value pass via command line doesn't show up as correct data type

wallacbe, 2015-12-05:
If I declare a config in the config file it shows up as int:
duration_s: 5
5 <type 'int'>

If I declare on command line with  "--config_value duration_s=5" it shows as str:
duration_s: 5
5 <type 'str'>

wallacbe, 2015-12-05:
Perhaps we can add ast.literal_eval when we are importing the args from command line?

fahhem, 2015-12-05:
I'm going on getting config to validate and convert values, you can rely on that instead. literal_eval won't catch putting the wrong type (such as "=5ms") and may convert values incorrectly (serial number of equipment might be all numbers). For now, I'd put int() around receiving a config value that I expect to be int because there's nothing stopping the config file from being wrong either.

wallacbe, 2015-12-05:
I also have a use case of passing a list.  For future would we assume passing just comma delimited values key=1,2,3  or key=[1,2,3]

fahhem, 2015-12-05:
I'd think the [], except that bash doesn't like that. What about specifying the flag multiple times? Or, if you're so pro config-via-command line flags, what about a flag that just takes yaml?

--config_yaml "key: 1, 2, 3\nother_key: yes"
--config_yaml "key: 1, 2, 3" --config_yaml "other_key: yes"

In the second case we'd merge the yamls and say the last value set wins in some order, such as "config file, config_yaml, config_value"

wallacbe, 2015-12-05:
Flag with yaml sounds good, then we'd get the same data types through command line as yaml file

jettisonjoe, 2015-12-09:
--config_yaml sounds like a really awesome idea to me.

closed 2016-09-20 by wallacbe

grybmadsci, 2016-09-20:
--config_value was never intended for production use, it's there for overriding values for testing.  --config_yaml is an interesting idea, but you're really just reinventing a heredoc:
```sh
python my_test.py --config_file=<(cat <<EOF
config_key: 5
EOF
)
```
still, i have no objection to the syntactic sugar :) i'm a little concerned about the order of precedence getting somewhat complex, perhaps we can disallow using --config_yaml *and* --config_value flags in the same commandline?  then it's clear that you have config file, and then maybe some overrides, instead of the hierarchy of precedence.