google/openhtf #72: phase_data raises "RuntimeError: Validate must only be called once" with loop=True

wallacbe, 2015-12-05:
To reproduce, add loop=True to test.Execute in hello_world.py and loop it twice:

sudo python hello_world.py --config hello_world.yaml 
Provide a DUT ID in order to start the test.
test
What's the widget type?
hi
Provide a DUT ID in order to start the test.
test
INFO:hello_world.py:Hello World!
What's the widget type?
hi
INFO:hello_world.py:Example says: Example Plug Did Stuff!
2015-12-05 15:51:42,619 - ERROR - Failed to execute test.
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 138, in _ThreadProc
    self._ExecuteTest(phase_executor)
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 164, in _ExecuteTest
    for phase_result in phase_executor.ExecutePhases():
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 150, in ExecutePhases
    result = self._ExecuteOnePhase(self._test_state.pending_phases[0])
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 180, in _ExecuteOnePhase
    result_wrapper.SetResult(phase_thread.JoinOrDie())
  File "build/bdist.linux-x86_64/egg/contextlib2.py", line 56, in __exit__
    next(self.gen)
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phase_data.py", line 147, in RecordPhaseTiming
    for name, measurement in measurement_map.iteritems()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phase_data.py", line 147, in <dictcomp>
    for name, measurement in measurement_map.iteritems()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/util/measurements.py", line 152, in Validate
    raise RuntimeError('Validate must only be called once')
RuntimeError: Validate must only be called once
ERROR:root:Thread raised an exception: Thread-1
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 138, in _ThreadProc
    self._ExecuteTest(phase_executor)
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 164, in _ExecuteTest
    for phase_result in phase_executor.ExecutePhases():
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 150, in ExecutePhases
    result = self._ExecuteOnePhase(self._test_state.pending_phases[0])
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 180, in _ExecuteOnePhase
    result_wrapper.SetResult(phase_thread.JoinOrDie())
  File "build/bdist.linux-x86_64/egg/contextlib2.py", line 56, in __exit__
    next(self.gen)
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phase_data.py", line 147, in RecordPhaseTiming
    for name, measurement in measurement_map.iteritems()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phase_data.py", line 147, in <dictcomp>
    for name, measurement in measurement_map.iteritems()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/util/measurements.py", line 152, in Validate
    raise RuntimeError('Validate must only be called once')
RuntimeError: Validate must only be called once
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 138, in _ThreadProc
    self._ExecuteTest(phase_executor)
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 164, in _ExecuteTest
    for phase_result in phase_executor.ExecutePhases():
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 150, in ExecutePhases
    result = self._ExecuteOnePhase(self._test_state.pending_phases[0])
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 180, in _ExecuteOnePhase
    result_wrapper.SetResult(phase_thread.JoinOrDie())
  File "build/bdist.linux-x86_64/egg/contextlib2.py", line 56, in __exit__
    next(self.gen)
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phase_data.py", line 147, in RecordPhaseTiming
    for name, measurement in measurement_map.iteritems()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phase_data.py", line 147, in <dictcomp>
    for name, measurement in measurement_map.iteritems()
  File "/usr/local/lib/python2.7/dist-packages/openhtf-1.0-py2.7.egg/openhtf/util/measurements.py", line 152, in Validate
    raise RuntimeError('Validate must only be called once')
RuntimeError: Validate must only be called once

amyxchen, 2015-12-08:
I can repro it, look at it.

fahhem, 2015-12-08:
Yes, I looked into it and the current state is that Measurement stores an
'outcome' from validation, so when you loop back around it still has that. It
seems to me that Measurement should be a write-once class with MeasuredValue
storing all the data from the use of measurements, such as the outcome.
If that's an acceptable plan-of-attack, I can work on that later today.

amyxchen, 2015-12-09:
I have a kind of heck quick fix to reset the outcome value to "None" each time creating a measurement_map. However, after talking to Joe, we agreed that the measurement should be persistent, and not reattach each time running the phases, and the outcome field should move to somewhere else and being updated there after each run of phases.

jettisonjoe, 2015-12-09:
I agree that MeasuredValue is probably where we want to store outcome information, since it's value-dependent. We should also set and validate at the same time in Collection's setattr.

There's an additional concern as well. _values are stored alongside _measurements in Collection currently. Which means we need to clear out each collection's _values at the end of each test run, unless we remove it from Collection and have it be kept and accessed under a test's record only.

fahhem, 2015-12-09:
Why not create a new Collection each time?

jettisonjoe, 2015-12-09:
Ah, I stand corrected. We already *are* creating a new Collection each time, but from the same set of measurements. So my concern about clearing _values was unwarranted. If we simply move outcomes out of Measurement and into MeasuredValue instead we should be okay. We might then also be able to take Measurement back down to a namedtuple so it's immutable.

grybmadsci, 2015-12-10:
Hm, this is code that I just authored a week ago, so I'd expected it to
potentially be a little buggy, I'll take a look.

Appreciate the help everyone.

fahhem, 2016-03-23:
madsci fixed this months ago

grybmadsci, 2016-03-23:
heck yeah i did! \o/

closed 2016-03-23 by grybmadsci