google/openhtf #79: Add default value option to conf Declare method

wallacbe, 2015-12-09:
Notes from HTF meeting:
Add default option to conf Declare method
provide a default station ID

jettisonjoe, 2015-12-09:
I see this as being particularly useful for plugs, as it seems many plugs are largely just managers of specific sets of config options. It's annoying if issue #60 can only be achieved for the most basic of tests. Having a way to specify default config values would definitely make invocations more user friendly for people coming up to speed on the usage of openhtf.

@grybmadsci, @fahhem, any strong feelings about this?

fahhem, 2015-12-14:
I think having a default would be very useful, many configs we've used are meant
to only be overridden in case the hard-coded value stops working.

grybmadsci, 2016-03-12:
pass `default_value=` to `conf.Declare()`

closed 2016-03-12 by grybmadsci