google/openhtf #80: Remove requirement for /var/run/openhtf directory

wallacbe, 2015-12-09:
Notes from HTF meeting:
remove /var/run directory since we have the multicast agent
remove mention of this from documentation

will simplify steps to run a station

jettisonjoe, 2015-12-09:
To elaborate a little, this change would make the multicast discovery service the single source of information about which stations exist on the network. Previously we had second layer of information because each station would also have access to its host's rundir, which might have runfiles from currently offline stations.

If we want to *reliably* display offline stations, I propose a solution where a config file or flag listing expected stations can be provided to the frontend. In production applications, where we know which stations we care about (and where deployment is likely automated through a tool) this would allow the frontend to serve as an at-a-glance indicator of station health. In non-production deployments, offline stations would only appear on the frontend's station list if they had previously been discovered through the multicast mechanism.

Then we can finally get rid of the use of physical disk as a crude IPC channel.

@grybmadsci, any thoughts?

closed 2016-03-10 by wallacbe

grybmadsci, 2016-03-10:
Oh, forgot to reply to this, but we talked about it a lot in person - Agree with removing /var/run/openhtf entirely.  At some point, we should add a way to explicitly add hosts that multicast might not find (depending on network configuration, the host may actually not be discoverable via multicast), but that can come later.

jettisonjoe, 2016-03-10:
@grybmadsci, yeah, I wanted to maybe do that in a frontend config file.