google/openhtf #90: measurement.py _AssertValidKey error

MasterChiefV, 2016-01-07:
With the latest code:

commit bace48a6bbf6f028a3a079a46e7761b80c015653
Merge: 256b4e0 91dacd6
Author: Joe Ethier <joeethier@gmail.com>
Date:   Tue Jan 5 13:03:44 2016 -0800
    Merge pull request #86 from fahhem/tox
    Use tox to test multiple environments

Following the measurement.py example to decorate a Measurement:

@measures(Measurement(
          'number_widgets').InRange(5, 10).Doc(
          '''This phase parameter tracks the number of widgets.'''))

I would get a repeated list of the following error:
File ".../openhtf/openhtf/util/measurements.py", line 315, in __getattr__
    self._AssertValidKey(name)
  File ".../openhtf/openhtf/util/measurements.py", line 300, in _AssertValidKey
    if name not in self._measurements:
RuntimeError: maximum recursion depth exceeded

jettisonjoe, 2016-01-07:
@fahhem, I'm guessing your changes to measurements.py introduced this? In particular it looks like accessing ```self._measurements``` on the mutablerecords version of Collection calls Collection's __getattr__, creating an infinite regress. I haven't looked into mutablerecords to investigate further yet.

fahhem, 2016-01-07:
Yeah, I actually had fixed this in mutablerecords when I sent the pull request.
However, I forgot that openhtf pins its versions tightly so I hadn't updated
setup.py. I have a PR to update it out now.

jettisonjoe, 2016-01-07:
This should be fixed by #91 which I just merged. Marking closed, but @MasterChiefV, please let us know if that didn't fix it.

MasterChiefV, 2016-01-08:
Just tried this out and I am still getting the same errors.

jettisonjoe, 2016-01-08:
Darn.

jettisonjoe, 2016-01-08:
Did you actually update your virtualenv by running setup.py develop after pulling the new codebase?

MasterChiefV, 2016-01-08:
Yup I forgot to do that. Sorry.  All is well now.  Closing this.

closed 2016-01-08 by MasterChiefV