google/openhtf #102: Add exit handler for plugs

wallacbe, 2016-01-24:
I'm adding a power supply plug.  I'd like to be able to exit the tester and be able to put the power supply in a known state.  I'm observing that TearDown() is called at the end of each phase.  Is there a way to only do a Teardown or a exit handler when testing ends regardless of state (PASS, FAIL or ERROR)?

Also an entry handler would be useful too

fahhem, 2016-02-13:
I had asked about this earlier and both grybmadsci and jettisonjoe said they didn't see a need for it. Could you explain your need a little more? Could your exit handler be a \__del\__ function?

fahhem, 2016-02-13:
(Forgot to mention, but putting something in \__del\__ can create complexity for the framework, so that may not be the solution either)

wallacbe, 2016-02-13:
Main use case is a power supply.  We want the power supply to be in a known state when the tester object is initially created, and when the DUT is done testing  The state could be OFF, 5V, etc...

We also don't necessarily want this tear down or exit handler state to happen in between phases.  We usually want the power to stay on between phases.

jettisonjoe, 2016-02-13:
I believe right now TearDown is only supposed to happen at the end of each
testrun, and that it's called as part of the executor's exit stack, so it
might just do what you want it to already. #117 might be preventing
TearDown from getting called when it's supposed to right now though.

wallacbe, 2016-02-29:
Issue #117

closed 2016-02-29 by wallacbe