google/openhtf #104: phase raising openhtf.util.measurements.InvalidType

wallacbe, 2016-01-29:
openhtf.util.measurements.InvalidType: Invalid measurement type: [Measurement(name='current_1', units=None, outcome=None, docstring=None, dimensions=None, validators=[]), Measurement(name='current_2', units=None, outcome=None, docstring=None, dimensions=None, validators=[]), Measurement(name='current_3', units=None, outcome=None, docstring=None, dimensions=None, validators=[])]

Trying to declare measurements with a loop:

@names.measures([names.Measurement('current_%s' % i) for i in ['1','2','3']])

fahhem, 2016-01-30:
Try this:

`@names.measures(*(names.Measurement('current_%s' % i) for i in ['1','2','3']))`

That creates an iterator and converts it into arguments

wallacbe, 2016-01-30:
That worked, thank you

Brandon Wallace | Test Engineer | wallaceb@google.com <lapetina@google.com>
*v**erily*

fahhem, 2016-02-13:
Did this work?

wallacbe, 2016-02-13:
Worked, closing issue

closed 2016-02-13 by wallacbe