google/openhtf #117: TearDown does not get called at the exit of test run

googlerhuili, 2016-02-10:
TearDown does not get called at the exit of test run.

I added raise Exception("TearDown") in the TearDown function. It is not raising the exception.

fahhem, 2016-02-12:
The framework catches TearDown exceptions and logs them, but to ensure that your station doesn't die right away, we pretend like everything is okay and try to continue:

https://github.com/google/openhtf/blob/a6deab3f721be856b6b55880ac56154dc24765ff/openhtf/plugs/__init__.py#L227

jettisonjoe, 2016-02-12:
Hui's description didn't entirely capture the issue; I was there when he discovered it. It wasn't that the exception wasn't raised, it was that the expected warning wasn't logged. We also tried adding some print debug to TearDown and then didn't see the expected printed output, which really does look like TearDown isn't actually getting called on plugs.

wallacbe, 2016-02-12:
I cloned the repo and added added a constructor with "print constructor",
added "print Tearing Down" and a deconstructor with "print deconstructor"

Provide a DUT ID in order to start the test.
test
constructor
constructor
What's the widget type?
asdf
Tearing Down
Tearing Down
deconstructor
deconstructor

I have an issue open for exit/entry handler already but this is kind of
related.

Could we have the plug object persist across a tester and use __init__ and
__del__ for the entry/exit handlers and have TearDown be what's done on
phase exit, and perhaps add a method that does something on Phase Entry?

Brandon Wallace | Test Engineer | wallaceb@google.com <lapetina@google.com>
*v**erily*

wallacbe, 2016-02-12:
I cloned the repo and added added a constructor with "print constructor", added "print Tearing Down" and a deconstructor with "print deconstructor"

Provide a DUT ID in order to start the test.
test
constructor
constructor
What's the widget type?
asdf
Tearing Down
Tearing Down
deconstructor
deconstructor

I have an issue open for exit/entry handler already but this is kind of related. 

Could we have the plug object persist across a tester and use __init__ and __del__ for the entry/exit handlers and have TearDown be what's done on phase exit, and perhaps add a method that does something on Phase Entry?

fahhem, 2016-02-13:
Let's not derail this issue for the other.

Hui, you put print statements in your TearDown and nothing got output? I'd be curious to see the output, since the only reason I see for TearDown not to get called is that your test didn't initialize the plugs in the first place. In which case, not tearing down plugs makes sense to me.

googlerhuili, 2016-02-25:
I just retried and the TearDown is called now.

Not sure what happened. But I created a new repo and cloned my committed
new codes. It works now.

grybmadsci, 2016-02-25:
@wallaceb: Joe and I talked about this a lot when we made the call to do
__init__ and TearDown(), it was kind of a tough call.  Ultimately, we
decided that the per-phase entry/exit is rare enough that it's okay to
require the test author to explicitly call the methods they want (and it
also hides less from non-author readers of the code, they don't have to go
hunting for these enter/exit methods to see plug side effects that may be
happening).

It's generally bad practice to use __del__ for this sort of thing because
you don't know when it's actually going to get called - in fact, there's no
guarantee it gets called at all, it's up to the garbage collector, which is
not specified as part of the Python language.  This is why we have
TearDown() instead of using __del__.  I'm open to reconsidering a per-phase
entry/exit, but it would take some convincing :)  In fact, I have a similar
use case at skybox, and what i did was implement __enter__ and __exit__ on
my plug to make it usable as a context manager, and then the first line of
my phase is `with camera:` and the rest is indented one level.  I like it
because it's clearer at a glance that it's happening, and it's only 1
additional line of code.

wallacbe, 2016-02-28:
Could we create plug methods for both phase and tester scope?

BringUpTester - Called when a tester is started or at the top of the tester loop
BringUpPhase - Called at the beginning of the phase
TearDownTester - Called at the end of the tester 
TearDownPhase - Called at the end of the phase

grybmadsci, 2016-02-28:
I don't really like having base class methods that you have to know to
override to get certain behavior.  The nice thing about __init__ is that
you *know* it's going to get called because the object has to get
constructed.  I don't even really like having TearDown(), but it's there by
necessity because __del__ is so bad to use in this case.

I also like having the visibility in the test phase of the with: block,
which only requires that you name your would-be BringUpPhase method
__enter__, and TearDownPhase becomes __exit__.  Having to include the with:
line at the top of every phase that uses the plug can get repetitive, I
admit, but I still prefer that over the failure mode of debugging
framework-invoked equivalents.

It's also a bit of a can of worms to have the framework do it, because what
do you do for timeouts?  is that included in the phase timeout, or is it
separate?  what do you do on error?  it creates code paths that are
annoying at best and can be very frustrating as a test author.  Is there a
reason for not wanting to use __enter__ and __exit__ besides the nuisance
of having to put the bulk of the phase code in a with: block?

~madsci

fahhem, 2016-02-28:
You could even avoid the extra indentation by using a decorator. If all
your phases do it, then you can even wrap them all at Test creation time.

I think the timeouts alone are a good enough reason to not make the
framework prescribe a way to do it.

wallacbe, 2016-02-29:
I synced from head and made a simple tester with plug.  I see plug's __init__ called at the beginning of each loop and TearDown called at the end (I also see TearDown when I added a rais in Phase2).  This behavior is actually what I'm requesting (unclear why my output earlier in this thread shows different).  I am seeing odd behavior of "Finishing test execution with state State.PASS."  getting appended each time.

python example.py 
Provide a DUT ID in order to start the test.
serial1
Plug __init__
phase 1
current: 1.0
phase 2
phase 3
current: 1.0
Finishing test execution with state State.PASS.
Plug TearDown
Plug __del__
Provide a DUT ID in order to start the test.
serial2
Plug __init__
phase 1
current: 1.0
phase 2
phase 3
current: 1.0
Finishing test execution with state State.PASS.
Finishing test execution with state State.PASS.
Plug TearDown
Plug __del__
Provide a DUT ID in order to start the test.
serial3
Plug __init__
phase 1
current: 1.0
phase 2
phase 3
current: 1.0
Finishing test execution with state State.PASS.
Finishing test execution with state State.PASS.
Finishing test execution with state State.PASS.
Plug TearDown
Plug __del__

import openhtf
import logging
import openhtf.plugs as plugs
from openhtf.names import *

class PowerSupply(plugs.BasePlug):
  def __init__(self):
    print 'Plug __init__'

  def TearDown(self):
    print 'Plug TearDown'

  def __del__(self):
    print 'Plug __del__'

  def read_current(self):
    return 1.0

@plug(ps=PowerSupply)
def Phase1(test, ps):
  print 'phase 1'
  print 'current: %s' % ps.read_current()

def Phase2(test):
  print 'phase 2'

@plug(ps=PowerSupply)
def Phase3(test, ps):
  print 'phase 3'
  print 'current: %s' % ps.read_current()

if __name__ == '__main__':
  test = openhtf.Test(Phase1, Phase2, Phase3,
      test_name='MyTest', test_description='OpenHTF Example Test',
      test_version='1.0.0')
  test.Execute(test_start=triggers.PromptForTestStart(), loop=True)

fahhem, 2016-02-29:
I filed #144 for the repeated output problem. So it seems like this issue is closed, right? The resource functions `__init__` and `TearDown` get called once per run and at the right times, and any phase entry/exit code can be put in `__enter__` and `__exit__` and used with `with` or a `@decorator`. We should document that somewhere.

PS, next time you post code, can you use the backtick (`) to quote it?

wallacbe, 2016-02-29:
SGTM

jettisonjoe, 2016-02-29:
Sounds like it's a bit beside the point now, but for the record, +1 to @grybmadsci's convention of using `__enter__` and `__exit__` and a big `with` block for per-phase contexts on plugs.

closed 2016-03-08 by jettisonjoe