google/openhtf #118: Add degrees per second and g's to UOM

wallacbe, 2016-02-11:
degrees per second (DPS)
g-force (g)

jettisonjoe, 2016-02-11:
Could have *sworn* DPS meant something else.

Anyway, I'm having a hard time finding an ANSI spec for either of those. Know what the "official" uom code would be for either of them?

wallacbe, 2016-02-11:
Hmm, yeah I don't see these in the ANSI spec.  degrees per second and g's were pulled from the module's datasheet.  Any thoughts on supporting non ANSI UOM?

fahhem, 2016-02-11:
There's radians per second, but nothing for g's (Newton's per kg, N/kg). However, there are degrees, seconds, newtons, and kilograms, so would we want to support 'units' that are multiples/divisions of each other?

wallacbe, 2016-02-12:
It's nice to be able to set limits directly from the datasheet without conversion.  Does mfg-insp require ANSI UOM?

jettisonjoe, 2016-02-12:
I'm not sure what mfg-inspector requires, but I'm open to having non-standard units in OpenHTF. Actually I think we used to have "metric sh*ttonne" in there and I took it out for open-sourcing, so I suppose we have an established history of allowing unofficial units.

fahhem, 2016-02-12:
Inspector does, right now, expect a single unit. However, its features are
driven by usage, so if OpenHTF drives that usage, then it can get into
inspector.

I think we can add the ability to have multiple units by creating a Units
class with the division and multiplication operators defined. Then when you
do UOM['DEGREES'] / UOM['SECONDS'], it's unambiguous and interfaces could
display the units trivially. What do you think?

wallacbe, 2016-02-13:
Does inspector require a ANSI UOM or would it accept a custom single unit like a "g"

jettisonjoe, 2016-02-13:
Isn't 'g' already a gram though (although not in our units.py)?

wallacbe, 2016-02-13:
I did some searching, it has a different symbol than the lower case 'g':
ɡ - ascii 0x261 - g-force
g - ascii 0x67 - gram
It can also be represented by 9.80665 m/s2 per ISO 80000-3

closed 2016-09-20 by wallacbe