google/openhtf #119: Build proto shouldn't be part of the regular build command.

jettisonjoe, 2016-02-11:
In theory, we're gonna have lots of users that don't use mfg-inspector someday. We shouldn't make them have to wrestle with protoc to use openhtf.

fahhem, 2016-02-11:
I don't think it is? It looks like he added the 'build_proto' command and in tox runs it. If you don't build_proto, you don't use protoc.

fahhem, 2016-02-11:
Oh, I didn't see line 83: `build.sub_commands.insert(0, ('build_proto', None))`

Removing that will make it work, but the errors may be confusing for anyone wanting to use the protos. There is the option of adding an 'extras' keyword in the `setup()` call that makes it so that someone can depend on `openhtf[mfginspector]` and they'd get all the extras needed for mfginspector, including building the protos.

grybmadsci, 2016-02-11:
Yeah, I think for now I prefer to have it in there as a sub command (in particular, our tests don't run without it).  The right solution is what farz mentioned with the extras keyword.  We'll have a lot of things that fall into this category, between output mechanisms and plugs.  Updated the issue to reflect this.

jettisonjoe, 2016-02-12:
Cool, I really like the "extras" idea. I'm excited to start implementing it. Frontend will also be an extra, I think.