google/openhtf #120: Testrun proto doesn't build on some linux systems.

jettisonjoe, 2016-02-11:
[snip from Brandon's email]
:~/source/openhtf$ python setup.py bdist_wheel
running bdist_wheel
running build
running build_proto
Attempting to build proto files:
./openhtf/io/proto/testrun.proto
./openhtf/io/proto/units.proto
google/protobuf/descriptor.proto: File not found.
units.proto: Import "google/protobuf/descriptor.proto" was not found or had errors.
units.proto:22:8: "google.protobuf.EnumValueOptions" is not defined.
units.proto: "google.protobuf.EnumValueOptions" is not defined.
testrun.proto: Import "units.proto" was not found or had errors.
testrun.proto:253:12: "Units.UnitCode" is not defined.
Traceback (most recent call last):
  File "setup.py", line 144, in <module>
    install_requires=requires,
  File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/dist-packages/wheel/bdist_wheel.py", line 175, in run
    self.run_command('build')
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/distutils/command/build.py", line 128, in run
    self.run_command(cmd_name)
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 70, in run
    ] + protos)
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/protoc', '--proto_path', './openhtf/io/proto', '--proto_path', '/usr/include', '--python_out', './openhtf/io/proto', './openhtf/io/proto/testrun.proto', './openhtf/io/proto/units.proto']' returned non-zero exit status 1
~/source/openhtf$ protoc --version
libprotoc 2.6.1

I also tried downloading the latest protoc: https://github.com/google/protobuf/releases

The system protoc symlink is pointing to /usr/local/bin/protoc.  Openhtf calls /usr/bin/protoc

which protoc
/usr/local/bin/protoc
:~/source/openhtf$ protoc --version
libprotoc 3.0.0
:~/source/openhtf$ /usr/bin/protoc --version
libprotoc 3.0.0

I''m able to get a the testrun_pb2.py  output if I manually do this:

:~/source/openhtf/openhtf/io/proto$ ls
__init__.py  testrun.proto  units.proto
:~/source/openhtf/openhtf/io/proto$ protoc testrun.proto --python_out=.
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
:~/source/openhtf/openhtf/io/proto$ ls
__init__.py  testrun_pb2.py  testrun.proto  units.proto
[/snip]

fahhem, 2016-02-11:
Seems like the descriptor.proto isn't in /usr/include/google/protobuf/ in that system, looks like "pkg-config --variable includedir protobuf" might tell us the proto_path to include instead of just blindly using /usr/include?

jettisonjoe, 2016-02-12:
@fahhem, you're right. A confirmed workaround for now is for users to simlink their proto stuff into the expected spots (/usr/bin and /usr/include).

fahhem, 2016-03-23:
Fixed by #122?

closed 2016-10-10 by jettisonjoe