google/openhtf #125: No mfg upload and looping ceases when phase raises exception.

wallacbe, 2016-02-12:
In hello_world.py I set loop=True, added mfg inspect key/user and callback, and had a phase raise an exception. Two issues observed:
1) Test stops looping
2) Test doesn't upload result to mfg inspector

python ./hello_world.py --config ./hello_world.yaml
Provide a DUT ID in order to start the test.
test
What's the widget type?
asdf
ERROR:root:Thread raised an exception: set_measurements
Traceback (most recent call last):
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 80, in _ThreadProc
    self._phase_result = self._phase(self._phase_data)
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/__init__.py", line 97, in __call__
    return self.func(phase_data, **plug_kwargs)
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/monitors.py", line 95, in MonitoredPhaseFunc
    return phase_func(phase_data, *args, **kwargs)
  File "./hello_world.py", line 70, in set_measurements
    raise FooException('Foo Exception')
FooException: Foo Exception
Exception in thread set_measurements:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/phasemanager.py", line 80, in _ThreadProc
    self._phase_result = self._phase(self._phase_data)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/__init__.py", line 97, in __call__
    return self.func(phase_data, **plug_kwargs)
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/monitors.py", line 95, in MonitoredPhaseFunc
    return phase_func(phase_data, *args, **kwargs)
  File "./hello_world.py", line 70, in set_measurements
    raise FooException('Foo Exception')
FooException: Foo Exception

DEBUG:hello_world.py:Finishing test execution with state State.ERROR.
2016-02-11 19:23:26,549 - ERROR - Failed to execute test.
Traceback (most recent call last):
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 138, in _RunTestLoop
    self.test.OutputTestRecord(self._test_state.GetFinishedRecord())
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/__init__.py", line 150, in OutputTestRecord
    output_cb(test_record)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/__init__.py", line 35, in __call__
    test_record).SerializeToString())
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 249, in TestRunFromTestRecord
    mangled_parameters = _ExtractParameters(record, testrun, used_parameter_names)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 170, in _ExtractParameters
    value = phase.measured_values[name]
KeyError: 'level_all'
ERROR:root:Thread raised an exception: Thread-1
Traceback (most recent call last):
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 113, in _ThreadProc
    self._RunTestLoop()
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 35, in _proc
    fn(*args, **kwargs)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 138, in _RunTestLoop
    self.test.OutputTestRecord(self._test_state.GetFinishedRecord())
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/__init__.py", line 150, in OutputTestRecord
    output_cb(test_record)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/__init__.py", line 35, in __call__
    test_record).SerializeToString())
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 249, in TestRunFromTestRecord
    mangled_parameters = _ExtractParameters(record, testrun, used_parameter_names)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 170, in _ExtractParameters
    value = phase.measured_values[name]
KeyError: 'level_all'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 113, in _ThreadProc
    self._RunTestLoop()
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/util/threads.py", line 35, in _proc
    fn(*args, **kwargs)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/exe/__init__.py", line 138, in _RunTestLoop
    self.test.OutputTestRecord(self._test_state.GetFinishedRecord())
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/__init__.py", line 150, in OutputTestRecord
    output_cb(test_record)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/__init__.py", line 35, in __call__
    test_record).SerializeToString())
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 249, in TestRunFromTestRecord
    mangled_parameters = _ExtractParameters(record, testrun, used_parameter_names)
  File "/home//venv/local/lib/python2.7/site-packages/openhtf-1.0-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 170, in _ExtractParameters
    value = phase.measured_values[name]
KeyError: 'level_all'

wallacbe, 2016-02-12:
I tried out the change, in mfg inspector I see "State.ERROR Code FooException" in Failure Reasons.  I compared it to some older test errors for another product and I saw Failure Reasons such as "FAIL , UsbWriteFailedError" and "TIMEOUT".  Do you think we can clean formatting for "State.ERROR Code FooException", Perhaps removing the "State.ERROR code"?

fahhem, 2016-02-12:
#129 should make the failure codes look better, want to give it a try?

PR #126 (2016-02-12): Handle missing measurement values in mfginspector output
Fixes #125
commit e04a0209 (2016-02-12): Handle missing measurement values in mfginspector output

Fixes #125
closed 2016-02-12 by grybmadsci