google/openhtf #127: Add support for AddOutputCallback configs

wallacbe, 2016-02-12:
We won't be using the same name/key for test run uploads for each tester instance.  Need way to select name/key.  When I do the following config isn't available at load time:

config = conf.Config()  
test.AddOutputCallback(output.UploadToMfgInspector(
      config.mfg_inspector_user,
      open(config.mfg_inspector_key, 'r').read()))

Raises this error:
openhtf.conf.ConfigurationNotLoadedError: mfg_inspector_user

fahhem, 2016-02-12:
You could use conf.ConfigValue('mfg_inspector_user') to get a function that will work, but UploadToMfgInspector still has to support it.

Alternatively, you could put a lambda around it and it'll work:

```py
config = conf.Config()

def UploadViaKeys(record):
  return output.UploadToMfgInspector(
      config.mfg_inspector_user,
      open(config.mfg_inspector_key, 'r').read())(record)

test.AddOutputCallback(UploadViaKeys)
```

fahhem, 2016-02-12:
I'm not saying the framework shouldn't handle it, but that will work until we decide how to handle this. `ConfigValue()` isn't a perfect answer since you can't `open()` it.

jettisonjoe, 2016-02-12:
Another approach for config-dependent upload behavior is to make a custom output module containing the conditional logic since config will be loaded by the time it's called. This output module can import and invoke other output modules conditionally.

fahhem, 2016-02-12:
That's what my code snippet above is, a custom output module. If the framework wanted to handle it, it would be an interesting problem due to wanting to do things like `open()` a config value. It seems like a 'custom output module' that's just a tiny function would be best. Mine ended up not being a lambda because all the names were long.

jettisonjoe, 2016-02-12:
Fair enough. I was more pointing out that you could still have a reusable separate 'module' rather than copy-pasting something like above in each tester's script. But perhaps that was already too obvious to everyone to need pointing out.

wallacbe, 2016-02-13:
what would be the syntax within the output module?  I'm seeing "openhtf.conf.ConfigurationNotLoadedError"

import openhtf.conf as conf
conf.Declare('token_port', default_value=443)
conf.Declare('dest_port', default_value=443)

class UploadToMfgInspector(object):  # pylint: disable=too-few-public-methods
  config = conf.Config()
  TOKEN_URI = 'https://accounts.google.com:%s/o/oauth2/token' % config.token_port
  SCOPE_CODE_URI = 'https://www.googleapis.com/auth/glass.infra.quantum_upload'
  DESTINATION_URL = ('https://clients2.google.com:%s/factoryfactory/'
                     'uploads/quantum_upload/' % config.dest_port)

grybmadsci, 2016-03-14:
I've actually removed the `ConfigValue()` mechanism as a means for config lazy-eval, I think it was always a bit hacky.  The way I'd suggest doing this is to make a custom output callback as follows:
```
@conf.InjectPositionalArgs
def ConfiguredUpload(test_record, mfg_inspector_user, mfg_inspector_key):
  with open(mfg_inspector_key, 'rb') as keyfile:
    mfg_inspector.UploadToMfgInspector(mfg_inspector_user, keyfile.read())(test_record)

if __name__ == '__main__':
  test = Test(...)
  # Lambda is required to pass record to ConfiguredUpload via keyword.
  test.AddOutputCallback(lambda test_record: ConfiguredUpload(test_record=test_record))
```

wallacbe, 2016-05-23:
I can access configs in callbacks, closing issue

closed 2016-05-23 by wallacbe