google/openhtf #131: Separate test initialization from execution

fahhem, 2016-02-13:
From the OpenHTF weekly meetings back in January, this was one of the things discussed. If there is still agreement on this, then I'll continue this work.

grybmadsci, 2016-04-04:
#132 went in quite a while ago, are there any more pending actions for this or can we close it?  if there are, can we get at least a high-level list of requirements to close this issue?

fahhem, 2016-04-04:
Yeah, the eventual future will be that a Test instance had a PlugManager, PhaseExecutor, TestExecutor, etc all setup in `__init__()` and ready to go in `Execute()`, which would allow a motivated user to replace one or more such pieces while keeping the interface intact to take advantage of everything else in OpenHTF.

Sorry about the ambiguity, I need to look at the code again to solidify it, which tends to end up with another PR for something else :)

fahhem, 2016-04-22:
@grybmadsci, I can't assign this to myself, can you assign it to me?

jettisonjoe, 2018-05-05:
This is done.

closed 2018-05-05 by jettisonjoe