google/openhtf #133: Shutting down is slow

fahhem, 2016-02-16:
It takes ~1 second for no good reason

fahhem, 2016-03-10:
I believe the causes for slow shutdown are the `join()` calls in the server `Stop()` methods. This is due to the way the main loop is told to exit:

https://github.com/google/openhtf/blob/262181734ceeb0649c38c28c4c0860ee0e32e46a/openhtf/io/frontend/__main__.py#L159
https://github.com/google/openhtf/blob/262181734ceeb0649c38c28c4c0860ee0e32e46a/openhtf/util/multicast.py#L88

The frontend sleeps for 5 seconds by default, and multicast sleeps for 3 seconds by default, so you can only exit on the next multiple of 5 seconds from when they were started.

fahhem, 2016-03-10:
For MulticastListener, there are two options, so I'd like to know which you prefer:

* Currently, it has a timeout of 3 seconds. This can be decreased to 1 without incurring much penalty, or even changed to a flag to allow someone to change it at will. The output unittest could specify 0.1 to allow it to exit quicker.
* Another way is to switch to using `select.select()`. This would only really help if we added a socketpair that we write to in `Stop()` to cause `select.select()` to return immediately. There is a [cross-platform way][1] to make a `socketpair()` that we can use.

The latter way would allow snappy shutdowns. Any other ideas? Suggestions?

[1]: http://code.activestate.com/recipes/525487-extending-socketsocketpair-to-work-on-windows/

grybmadsci, 2016-03-10:
I believe that select() on windows works with sockets (just not files), so
switching to select() should be fine, with no need to make custom wrappers.

grybmadsci, 2016-03-10:
Ah, i see the problem, i looked at it too quickly the first time... that's
a little ugly, the fallback to TCP loopback, but i suppose it's not that
bad, i've got no objection to adding a helper to do the socketpair business
and then using that to stop the multicast listener immediately.

Any thoughts, Joe?

~madsci

jettisonjoe, 2016-03-14:
I'm down with the socketpair approach, and it seems better than an adjustable timeout.

jettisonjoe, 2016-03-24:
So it turns out the multicast arrangement we're using is flawed. When running multiple instances of the framework on one machine, only one can bind to the multicast port, which hurts for server-based execution as well as pseudo multi-up uses. I think I'm going to switch it so that the framework periodically advertises itself using a multicast send, and the _frontend server_ binds to the multicast port and listens for advertisements. This results in more spam on the network, but re-enables multiple framework instances on top of one networking stack.

This would also simplify shutdown from the framework side, but we still might want to do the socketpair thing in the frontend server.

Also, it feels weird to say frontend server. Should we call it something else like "gui server" instead, or am I the only one who thinks it sounds kinda weird as-is?

grybmadsci, 2016-04-04:
Still going to leave this open, waiting on a cross-platform socketpair solution right?

fahhem, 2016-04-05:
In another issue, @grybmadsci brought up moving the Multicast/HTTP server into `Initialize` instead of in `Execute`, which I support. This would decrease the impact of the `select` replacement, do we still care if unittests aren't affected? This bothered me within the tests only.

jettisonjoe, 2016-04-05:
I agree it's less of an issue outside of the tests. At the same time, there's an unfiled issue where if the machine doesn't have link-layer connectivity on any interfaces (or loses it partway through), the multicast listener barfs and the frontend never finds that station again. I was going to use the socketpair PR as a time to fix that bug as well, but I'll file it as a separate issue so we can close this one.

grybmadsci, 2016-04-05:
All right, I think we can close this one out then - as a side note I also intend to make `Initialize()` take a couple framework-tuning knobs as arguments, such as the ability to disable the `http_api`, which can be used in tests to lower the overhead of unit tests that want to execute an end-to-end test but don't care about the `http_api` being up.

closed 2016-04-05 by grybmadsci