google/openhtf #140: Skip attachments with JSON output module

wallacbe, 2016-02-23:
Using JSON output module with binary output file raises:
UnicodeDecodeError: 'utf8' codec can't decode byte 0x89 in position 0: invalid start byte

Propose we skip attachments for this module or skip the encoding and just save binary file?

grybmadsci, 2016-02-23:
There's an option to skip attachments already, pass inline_attachments=False when you create the OutputToJSON object.

That said, we really need to fix this for real - we should check the MIME type, and anything but plaintext should get uuencoded when we serialize.  We might also consider putting a size limit (or at least a warning when you go over a certain size) to avoid gigantic json files blown up even larger with huge uuencoded binary data that should really just get spit out in a standalone file.

grybmadsci, 2016-03-23:
Can't assign @fahhem for some reason, but he's poking around at similar things and might fix this along the way.. otherwise I'll do it eventually :)

glados-verma, 2022-10-12:
Please reopen if still applicable.

closed 2022-10-13 by glados-verma