google/openhtf #141: add config values to default output modules for added traceablity

wallacbe, 2016-02-23:
add config values to default output modules for added traceablity

grybmadsci, 2016-02-24:
you could do this manually pretty easily by doing config=conf.Config().dictionary in the constructor of the Test object, and it would show up in metadata.

what are everyone's thoughts on doing this vs. implicitly adding config to metadata vs. creating a 'config' top level key in the test record and putting it there?  any of these are fairly easy to do.

wallacbe, 2016-02-24:
+1 implicitly add to metadata

wallacbe, 2016-06-04:
Looks like config=conf.Config().dictionary is no longer supported.  Should I add a GetDictionary method to Configuration class?

grybmadsci, 2016-06-05:
nope, use `conf._asdict()`, already done for you ;)

wallacbe, 2016-06-05:
Thanks that worked.  Any qualms about accessing private attribute, should I change to public?

grybmadsci, 2016-06-05:
nope, it's not actually private, it's underscored to avoid name collision (config attributes can't start with underscore for that reason).  we also established a convention internally that `_asdict()` is used to provide serialization-friendly dictionaries of things, so it's consistent with that.

closed 2016-06-05 by wallacbe