google/openhtf #142: Console logs inconsistent with test_record logs (console missing lines)

grybmadsci, 2016-02-24:
I've got a few test phases, and for some reason calling test.logger.info(...) isn't producing console logs until partway through the test, when they start appearing.  All log lines are in the output test_record, so i'm guessing it's something to do with log levels to console not being set up correctly?  i'll look into it more later and see if i can't figure out what's triggering the change, might lead to the cause of the problem.

amyxchen, 2016-02-24:
test.logger only log openhtf framework info, not user test info. to setup console log, use flag '--verbosity LEVEL'. LEVEL is one of ['debug', 'info', 'warning', 'error', 'critical'], default is 'warning'], default is 'warning'

amyxchen, 2016-02-24:
test.logger only log openhtf framework info, not user test info. to setup console log, use flag '--verbosity LEVEL'. LEVEL is one of ['debug', 'info', 'warning', 'error', 'critical'], default is 'warning'.

grybmadsci, 2016-02-24:
hm, right, but that wouldn't explain why some logs are showing up and some
are not.

jettisonjoe, 2018-05-05:
This is fixed with the recent logging refactor.

closed 2018-05-05 by jettisonjoe