google/openhtf #144: TestState initialized logging every time

fahhem, 2016-02-29:
Found in #117, because `logger.addHandler` calls being repeated in a looped test, there are multiple problems:

* There's multiple StreamHandlers on all subsequent test executions
* The previous run's RecordHandler remains on the logger.

The latter doesn't cause noticeable effects because the records are outputs once, but it creates an O(n<sup>2</sup>) memory leak.

wallacbe, 2016-03-06:
In class TestState(object)
I added:
def __del__(self):
    self.logger.handlers = []

and I don't see the repeated output.
Is this an acceptable approach to fixing this?

wallacbe, 2016-03-06:
In class `TestState(object)`
I added:
`def del(self):
    self.logger.handlers = []`

and I don't see the repeated output.
Is this an acceptable approach to fixing this?

grybmadsci, 2016-03-06:
hm, that's a solution, but not entirely complete.  the problem is that we're mis-using Python's logging module in test_state - you're not supposed to set up loggers multiple times, and TestState is written to fully set up the logger each time one is created.

we should move the set-up (everything but the addHandler() call with the RecordHandler) of the RECORD_LOGGER logger to setup_logger() here:
https://github.com/google/openhtf/blob/master/openhtf/util/logs.py#L137

And then we should create the RecordHandler and save a reference to it on the test_state object, so that you can removeHandler() it in __del__ and also in GetFinishedRecord at:
https://github.com/google/openhtf/blob/master/openhtf/exe/test_state.py#L134

Because GetFinishedRecord is called at the end of the test, and __del__ is unreliable.  Calls to removeHandler() with a handler that is already removed are just a no-op, so it doesn't hurt to remove it in both places.

grybmadsci, 2016-03-14:
Thanks @wallacbe !

closed 2016-03-14 by grybmadsci