google/openhtf #145: Convert AddOutputCallback(self, callback) to AddOutputCallback(self, *callbacks)

wallacbe, 2016-03-01:
Would it be worthwhile to convert AddOutputCallback(self, callback) to AddOutputCallback(self, *callbacks) so we can add multiple callbacks in one line or pass in callbacks as *args (similar to phases)?

jettisonjoe, 2016-03-14:
Seems like a good idea to me.

grybmadsci, 2016-03-17:
Do we want to keep `AddOutputCallback()` and add the new one, or remove the old one and update all existing tests?

jettisonjoe, 2016-03-24:
Because the new name would be `AddOutputCallbacks` (with an s), right? Per our verbal conversation last weekend, I think our general practice should be to log a deprecation warning when using the feature on the release before the one that removes it. Since we're not versioned or in pypi yet, whether we do that in this case is a toss-up.

closed 2016-04-20 by grybmadsci