google/openhtf #146: OSX Support

wallacbe, 2016-03-02:
Some folks are trying to run on OSX.  Documenting issues
had to homebrew build SWIG
m2crypto was causing issues 

brew install openssl
 brew install swig
 sudo chown <user> /usr/local/bin
 brew link swig
 brew lin pcre
 sudo chown root /usr/local/bin
 sudo chown <user> /usr/local
 brew update
 sudo chown root /usr/local

wallacbe, 2016-03-02:
m2crypto actually still causing issues (adb usb plug)

jettisonjoe, 2016-03-14:
I'll try to take a look at this this week.

grybmadsci, 2016-03-14:
The 'Windows' support I did actually also includes OSX support, I believe.
Do you need to use the ADB USB plug?  If not, you should be able to use
just the openhtf package, without the usb plugs extra, and not depend on
M2Crypto.

Thanks for documenting the brew steps, though, that's super handy and we
should roll it into the docs!

wallacbe, 2016-03-14:
Don't need to use ADB USB plug.  How do we configure  to not use usb plugs extra?  Thanks

grybmadsci, 2016-03-14:
Hm, you may need to perform the build step on a linux machine, but the
install process should simply not depend on M2Crypto unless you explicitly
pip install openhtf[usb_plugs].  Try `python setup.py sdist` on a linux
machine, grab the .tar.gz from dist/, and then `pip install <file>` that on
OSX?

Sorry, I'm not too familiar just yet with some of the nuances of python
packaging and dependencies, this is just some of the flailing that I've
done, so maybe it helps.

grybmadsci, 2016-04-20:
I've successfully run openhtf tests on my Mac now (with usb plugs moved into an extra, those haven't been made to work).  @wallacbe can you give it a go from HEAD and let me know if you encounter any issues?  at some point we should get a clean mac image and document all the brew packages that need to be installed (i don't think there's many) alongside the apt-get equivalents.

wallacbe, 2016-06-09:
I have a fresh mac that I'm setting up and will update CONTRIBUTING.md.  Go ahead and assign issue to me.

wallacbe, 2016-06-24:
I was able to build and run hello world on MacOs after these steps:

# Install Xcode

# Install Homebrew Mac

brew install pkg-config libffi

brew install protobuf

git clone https://github.com/google/openhtf.git

sudo easy_install pip

# Make sure pip is up-to-date.
sudo pip install --upgrade pip

# Install virtualenv via pip.
sudo pip install virtualenv

# Change to the openhtf directory.
cd openhtf

# Create a new virtualenv.
virtualenv venv

# Activate the new virtualenv.
. venv/bin/activate

maddychan, 2016-06-27:
The steps above work fine, now trying to get it to python setup.py develop smoothly. Doesn't go smoothly now because El Capitan took out ssl headers, tried a couple different commands to try to fix it but haven't worked yet. It's possible that it requires a Xcode command line tools update so trying that now, update is taking a while.

jettisonjoe, 2016-07-19:
This is done!

closed 2016-07-19 by jettisonjoe