google/openhtf #147: Add green/red text for Pass//Fail Status in CLI

wallacbe, 2016-03-02:
Add green/red text for Pass//Fail Status in CLI.  Allows us to see test result better if front end isn't running

amyxchen, 2016-03-02:
I am on this.

jettisonjoe, 2016-03-03:
For reference, the plan we talked about was to implement CLI test results as another output module you can optionally add to your tests. Does that sound reasonable?

grybmadsci, 2016-03-04:
Hm, yeah sounds pretty reasonable, but depending on how much you plan to make it look pretty, this is actually a giant can of worms... even something as seemingly innocuous as red/green text is actually not only OS-specific, but terminal-type specific.  You might consider using a library like ncurses to help, but it'll still be very tough to make truly cross platform and friendly.

jettisonjoe, 2016-03-04:
Agreed cross-platform colorization is going to be a can of worms for sure. I was thinking CLI test results are important enough that we should try to put something out there with best-effort colorization and refine it later.

wallacbe, 2016-06-02:
colorama 0.3.7 may be an option here.
Cross-platform colored terminal text.

wallacbe, 2016-09-20:
Output module exists for this.  Colorama tracked in https://github.com/google/openhtf/issues/300

closed 2016-09-20 by wallacbe