google/openhtf #148: Add option to backup mfg_inspector.py pb file on local filesystem

wallacbe, 2016-03-02:
Add option to backup mfg_inspector.py pb file on local filesystem

wallacbe, 2016-03-06:
hello world example is using json file extension for proto output, thus overwriting first json that was created.  Propose we change hello_world example to use .pb as the file extension for more clarity  

`test.AddOutputCallback(json_factory.OutputToJSON(
      './%(dut_id)s.%(start_time_millis)s.json', indent=4))
  test.AddOutputCallback(output.OutputToTestRunProto(
      './%(dut_id)s.%(start_time_millis)s.json'))`

fahhem, 2016-03-06:
This fix was part of #132, which has stalled in code review for over 3 weeks now. If that hadn't stalled for so long, this would be fixed by now. Feel free to create a new PR to fix just that, hopefully they'll review it faster than mine.

grybmadsci, 2016-03-06:
#132 will get merged when the changes requested by the project author have been made.

closed 2016-03-06 by grybmadsci