google/openhtf #149: Establish best practice for handling multiple SKUs

wallacbe, 2016-03-02:
One example is a product that has multiple colors and is sharing a test phase and requires unique limits for a measurement based on SKU/color.

wallacbe, 2016-03-02:
Can measurements be dynamically declared at run time?

jettisonjoe, 2016-03-14:
The way it stands, measurements can't be easily declared within a phase. But you _might_ be able to add validators to them within a phase.

grybmadsci, 2016-03-14:
I believe farz's recent change allows you to reuse phases with different
@measures decorators.

jettisonjoe, 2016-03-14:
Sounds like a prime thing to have an example of in examples/. #166

glados-verma, 2022-10-12:
A phase is simply a function decorated with one or more OpenHTF decorators that convert it into a phase. One mechanism we've used is to call the decorators rather than use the decoration "syntactic sugar". I'll add an example showing that this is an option.