google/openhtf #155: Convert InRange to enhanced string argument in Measurement

wallacbe, 2016-03-08:
Get rid of InRange method and move to string argument in Measurement().
Add support for less/greater equal than comparator.  Discussion in meeting was to add a string argument to Measurement and define comparison as follows:
'[x, y]'
'(x, y]'
'(x, y)'
'[x, y)'
And unbounded would look like
'[,y]'  or potentially '[None, y]'
'(x,]' or potentially '(x, None]'

fahhem, 2016-03-08:
How about using normal comparison operators? So '1 > x > 3', 'x < 3', or 'x == 3'. Using the builtin module `parser`, we can determine the symbol used (so they don't have to use a specific one), and you can check that it's an expression. We can also provide extra symbols, so if we want to provide math.sin and friends, it's easy.

grybmadsci, 2016-03-08:
+1 to more Pythonic range expressions.  Also, I'd like to keep the InRange() convention in addition to parsing string constraints, as it's a little more friendly to people that are more comfortable with a programmatic interface (and that mechanism is much more extensible for complex validators, such as might be used for multidimensional data).

fahhem, 2016-03-08:
Though I just realized that getting the actual comparison numbers out of the expression can be difficult, so if you put '1 < x < 3', then reporting the 1 and 3 can be a hassle. It's possible to 'optimize' for certain common expressions, but things can get out of hand easily if someone does '1 < x and sin(x) < 3'. If they use an expression, I'd just output the string itself to json/proto/inspector/etc

fahhem, 2016-03-08:
If we use `sympy`, we can handle '1 < x' and get the 1 and x part really easily. Unfortunately, we lose the ability to do multiple comparisons, so 'x < 1 and x < 3' is a parse error (no matter the form of and, &, or &&).

jettisonjoe, 2016-03-08:
Personally I like the comma-style interval notation, which conveniently limits the types of comparisons that can be expressed to only ranges, which is what the InRange validator is all about. Having the interface _appear_ more Pythonic suggests more flexibility than I was imagining us building into InRange, possibly misleading test authors to think that any Python expression might work.

But ultimately I'm open for whichever of these syntax choices test authors find more intuitive and better to work with.

I agree we should keep the normal numerical arg option as well, in addition to the string parsing option.

fahhem, 2016-03-09:
Do we have a list of test authors to poll? We could use an external email address for this sort of thing

grybmadsci, 2016-03-09:
I'm running through the mental exercise of what exactly this would look like, and also trying to understand the motivation for this change.  It sounds like the intention is to come up with a shorter-hand way of specifying measurement validators, and potentially less confusing to people unfamiliar with OO paradigms than the .Foo().Bar().Baz() way we currently force.

If that's the case, then I'd imagine it would look something more like:
```
# Specify a numeric measurement with string-specified limits.
@measures('foo', limits='(1, 10]', doc='This is a foo measurement')
# Maybe specify a numeric measurement with non-string limits?
@measures('bar', limits=(5, 10))
# Specify a string regex to match
@measures('baz', regex='.*Baz')
# Specify a custom validator
@measures('harbl', validator=lambda x: 1 < x ** 2 < 10)
def TestPhase(test):
  pass
```

I think for the target audience in mind, this makes more sense than the single `@measures` with multiple `Measurement` objects passed in - the extra levels of indirection there are confusing.  It wouldn't be too hard to overload `@measures` to check the type of the first arg and fork between two ways of interpreting the args.

My thought here is that the ability to pass a lambda like this fits the need for more complex Python expressions, while `limits` and `regex` keys fit the two 99%-of-validators cases.

I left out units and dimensions, but one could easily see them being kwargs to `@measures` also.  On the implementation side, you check the first arg of `@measures` for basestring, and if it is, you construct a single Measurement based on that string and the remaining kwargs and do the same as if the user had done the full `@measures(Measurement('foo').Doc('bar').InRange(...))` syntax.

This is all just throwing ideas out there, I'm not super attached or opinionated about this in general, so long as we don't entirely remove the ability to use custom validators (which are going to be essential for multi-dimensional analysis).

jettisonjoe, 2016-03-09:
@grybmadsci, Funny you should mention the chaining syntax, as that came up sort of separately in the same meeting. It seems like people are finding it cumbersome and confusing to work with. I , too, proposed keyword arguments to the Measurement constructor as an alternative at the meeting, and @wallacbe for one seemed down with the idea. In particular, I think test authors would find it nice to be able to put the args on separate lines for readability.

Based on the meeting discussion, I think the motivation is twofold. Primarily, people aren't liking the chaining syntax, and secondarily, the current InRange() doesn't give full control of in-/exclusivity, and offsetting the limit by a small "epsilon" seems hacky.

I like your imagined solution a lot; it's nearly indistinguishable from what I had in mind myself.

jettisonjoe, 2016-03-09:
@fahhem, I don't have a list, per-se. We can poll our respective groups internally and/or advertise the discuss list you recently created. That weekly meeting is supposed to be the internal real-time forum for discussing this stuff, and github is supposed to be the external forum for it, as we're doing here.

wallacbe, 2016-07-22:
Thoughts on adding a new validator to do (Max, Min) ?  InRange currently does [Max, Min]

fahhem, 2016-07-23:
I was looking at voluptuous recently and it looks like it has all our validators and more and even a better range check that lets you specify if the actual min and max values are included. Previously, voluptuous used nested functions to do validation, which meant we couldn't get the values out. But that changed in the past few months to be classes so we can use it now!

Here's their Range: https://github.com/alecthomas/voluptuous/blob/master/voluptuous/validators.py#L447

If you specify `min_included=False, max_included=False` you'd get your (Min, Max) validator.

wallacbe, 2016-09-20:
Closing, track in https://github.com/google/openhtf/issues/415

closed 2016-09-20 by wallacbe