google/openhtf #156: Setting unbounded measurement to None results in fail

wallacbe, 2016-03-09:
Observed that setting unbounded measurement to None results in State.FAIL status.  Is this expected behavior?

wallacbe, 2016-03-10:
https://github.com/google/openhtf/issues/4 appears to be related to this

jettisonjoe, 2016-03-10:
I think we had tabled this with the intention of doing something sensible when we move measurement validation to the time of setting. Fixing #4 should fix this.

grybmadsci, 2016-04-27:
Updates to measurements have included a separate flag to track whether a measurement has been set or not, rather than using None to indicate "not set".  So now setting to None is a valid thing.  Unset measurements still cause the test to FAIL (we can discuss that separately), but setting a measurement to None no longer makes it revert to UNSET.

closed 2016-04-27 by grybmadsci