google/openhtf #161: Considerations for using plugs in other plugs.

grybmadsci, 2016-03-10:
Just tossing some ideas around, I know we explicitly wanted to avoid a complex dependency system.  I'm currently facing a situation where I have a handful (4-5) low-level device plugs, and a 'fixture' class that pokes at all of them and provides a higher level interface for test code.  I could have each phase that wants to use this fixture instantiate a new fixture and pass in the plugs, which is the current paradigm for such a thing.

I've gotten very used to using `@conf.InjectPositionalArgs`, and I've found it to be a paradigm that I really like.  So I was thinking from a user standpoint, `@plugs.InjectPositionalArgs` would be incredibly handy, so it would look something like this:
```
import openhtf.conf as conf
import openhtf.plugs as plugs

conf.Declare('my_plug_port', 'port my plug is connected to')

class MyDevice(plugs.BasePlug):
  @conf.InjectPositionalArgs
  def __init__(self, my_plug_port):
    self.port = my_plug_port

class MyFixture(plugs.BasePlug):
  @plugs.InjectPositionalArgs(my_device=MyDevice)
  def __init__(self, my_device):
    self.device = my_device
```

Looking at the implementation of `conf.InjectPositionalArgs`, you could even use both decorators on the same constructor because they pick out positional args by name and then pass them as **kwargs.  I really like the user-facing look of this, but the implementation under the hood gets quite complex.

At a glance, it would mean annotating plugs with dependencies (the decorator could stash this on the decoratee), and then recursively walking those dependencies when building the plug type map.  This could actually be fairly easily implemented on BasePlug, since we require all plugs to subclass it, making the recursive walk straightforward enough.

Then we'd have to update the actual building of the plug_type_map here:
https://github.com/google/openhtf/blob/17c50d148e92bc23300c08e88b1be192c7b3bb77/openhtf/__init__.py#L131
To incorporate plugs' dependencies.  Now that I write it all out, it's not *that* bad.  What do you guys think?  I'm normally the one pushing back on complex features like this, but I think the user-facing side is clean enough and the implementation is just barely not-too-complex enough for me to like it.

fahhem, 2016-03-10:
The point of the injecting (at least for my needs) is to be able to replace the lower level plugs with others, right? Such as saying "I need a power supply" and then at some point (at `Test()` creation, `Execute()`, config, etc) someone goes "oh, this is the power supply you have".

Otherwise, you can just call `MyDevice()` in `MyFixture()`, right? Or is this to allow the test phases access to the same `MyDevice()` instance as `MyFixture()` has?

In either case, I wouldn't call them both `InjectPositionalArgs()`, `plugs.InjectPlug()` or similar would be easier to understand. But yeah, the API as-is is pretty simple, at least compared to the `Provide()/CanProvide()/DoneWith()` triplet I had previously.

My other need around plugs is the ability for a plug to be 'split up' into multiple, such as a dual-channel power supply being split into two 'normal' power supplies or the DUT fixture into each MCU's comms channel. This allows the test to use either power channel, without depending on the exact physical hardware setup. And for the DUT, you can use the same phase with different MCU's without having to write it or a wrapper function to 'split up' the resource each time.

grybmadsci, 2016-03-10:
I considered simply instantiating the child plug in the parent one, but then you don't automatically get the TearDown() call.  That said, you can just do it yourself in the parent TearDown(), so perhaps it's better to stick with that instead of introducing new concepts.  I think using the same instance of a child plug directly in a phase is an uncommon use case, and you're right about that being the only significant delta, the rest is just boilerplate in the parent plug.

jettisonjoe, 2016-03-10:
As much as I'd like to push the '@receptacle' concept to mate with plugs in plugs, I'm wondering how often it's the case that we actually realize value from the flexibility of injection rather than of just hand-instantiating the needed sub-plugs and writing the needful bit of boiler plate.

If we think it's valuable, then we also should consider where that leads; do we need to provide base classes for generic test equipment types like "PowerSupply" that all plugs must subclass, along with type checking in the injection mechanism? How widely to different power supplies differ, and what do we do when the power supply I bought is missing a feature needed by the superplug?

We were starting to go down this road with the EtherSync (remote USB) plug, but tabled the issue when we switched focus to distributed rather than centralized test execution.

As for giving the phase code access to the same instance, isn't the subplug gonna be an attribute of the superplug anyway? The only difference I see is that subplugs are locked to a specific type, and are not managed by the framework but by their superplugs directly. To get around the type thing I guess you could build a compatibility layer that wraps your power supply and subclasses a different brand's power supply, but that just seems messy and there are too many possible permutations of 'need this, have that'.

I could be convinced either way on this, but right now I'm leaning towards not introducing new concepts.

grybmadsci, 2016-03-10:
This ends up being a can of worms, because the syntax I had in mind is as complex as I'd really want to consider, and it's not complex enough to support true injection (it's not configurable externally, you still have to edit the code directly to pick a different thing to inject).  I'm inclined to agree with what I think is the general consensus that while it's "neat" and "doable", it's probably just on the wrong side of the can/should line, tabling the idea for now :D

grybmadsci, 2016-03-12:
Joe and I talked about this a bit more, and the paradigm we'd like to go with for now is that you instantiate the 'superplug' in a phase, and if you need it to persist across phases, you can stash an instance in test.state.

A side issue that came up is if you want to instantiate the same plug but with different configs (ie, you have two PSUs, each on a different serial port).  There paradigm we'd like to encourage here is that you can subclass the PSU plug, decorate __init__ with a more semantically meaningful config key (like camera_psu_port, instead of just agilent_port), pass that config key into the base class's __init__, and then use that subclass in your test phase instead of the base class:

In agilent.py or something similar:
```
conf.Declare('agilent_port')

class AgilentPSU(plugs.BasePlug):
  @conf.InjectPositionalArgs
  def __init__(self, agilent_port):
    self.port = agilent_port
```

In my_fixture.py or something more descriptive:
```
conf.Declare('camera_psu_port')
conf.Declare('chamber_psu_port')

class CameraPSU(agilent.AgilentPSU):
  @conf.InjectPositionalArgs
  def __init__(self, camera_psu_port):
    super(CameraPSU, self).__init__(agilent_port=camera_psu_port)

class ChamberPSU(agilent.AgilentPSU):
  @conf.InjectPositionalArgs
  def __init__(self, chamber_psu_port):
    super(ChamberPSU, self).__init__(agilent_port=chamber_psu_port)
```

And then in your test:
```
@plugs(camera_psu=my_fixture.CameraPSU,
             chamber_psu=my_fixture.ChamberPSU):
def MyTestPhase(test, camera_psu, chamber_psu):
  pass
```

This adds a little boilerplate overhead, but we think it makes the test code more explicit and clear what's going on.  Closing the issue, I think we've concluded there's no code changes required.  With updates to examples to encourage these paradigms (separate issue already created), I think we'll be good.  Thanks guys for the thoughts/comments!

closed 2016-03-12 by grybmadsci