google/openhtf #163: frontend raises error in fresh virtual env since station_list.html isn't present

wallacbe, 2016-03-11:
I have a py_binary (that contains the virtual env) that launches different test types based on a flag.  I am trying to launch the frontend from within the py_binary before running the tester.  I can't run openhtf.io.frontend.main() so I am launching it like this:
`path = sys.executable`
`os.system('%s -m openhtf.io.frontend' % path)`

I run the py_binary and see the frontend server come up and I go to http://localhost:12000 and see "500: Internal Server Error" instead of scan box and then the following error in the CLI.  Looks like station_list.html isn't present. 

Starting openhtf frontend server on http://localhost:12000.
ERROR:tornado.application:Uncaught exception GET / (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='GET', uri='/', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, sdch', 'Host': 'localhost:12000', 'Accept': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.87 Safari/537.36', 'Connection': 'keep-alive', 'Upgrade-Insecure-Requests': '1'})
Traceback (most recent call last):
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/openhtf/io/frontend/__main__.py", line 200, in get
    self.render('station_list.html', stations=self.store.stations)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/web.py", line 699, in render
    html = self.render_string(template_name, **kwargs)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/web.py", line 803, in render_string
    t = loader.load(template_name)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/template.py", line 424, in load
    self.templates[name] = self._create_template(name)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/template.py", line 451, in _create_template
    with open(path, "rb") as f:
IOError: [Errno 2] No such file or directory: '/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/openhtf/io/frontend/templates/station_list.html'
ERROR:tornado.access:500 GET / (127.0.0.1) 1.66ms
ERROR:tornado.application:Uncaught exception GET / (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='GET', uri='/', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, sdch', 'Host': 'localhost:12000', 'Accept': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.87 Safari/537.36', 'Connection': 'keep-alive', 'Cache-Control': 'max-age=0', 'Upgrade-Insecure-Requests': '1'})
Traceback (most recent call last):
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/openhtf/io/frontend/__main__.py", line 200, in get
    self.render('station_list.html', stations=self.store.stations)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/web.py", line 699, in render
    html = self.render_string(template_name, **kwargs)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/web.py", line 803, in render_string
    t = loader.load(template_name)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/template.py", line 424, in load
    self.templates[name] = self._create_template(name)
  File "/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/tornado/template.py", line 451, in _create_template
    with open(path, "rb") as f:
IOError: [Errno 2] No such file or directory: '/tmp/tmp.zeHhvvaTqL/virtualenv/lib/python2.7/site-packages/openhtf/io/frontend/templates/station_list.html'

jettisonjoe, 2016-03-14:
Hmm. How is the binary being built?

fahhem, 2016-03-14:
We may need a MANIFEST.in for getting non-python files into wheels?
https://github.com/chainreactionmfg/cara/blob/master/MANIFEST.in

wallacbe, 2016-03-14:
built with bazel then zipped up into py_binary

wallacbe, 2016-03-18:
This time I used the virtual env method described in the documentation, built and installed openhtf in the virtual env and added the following to hello_world (above test = Test(hello_world)

` import sys, os`
` path = sys.executable`
 `os.system('%s -m openhtf.io.frontend' % path)`
 `test = Test(hello_world)`

Starting openhtf frontend server on http://localhost:12000.
ERROR:tornado.application:Uncaught exception GET / (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='GET', uri='/', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, sdch', 'Host': 'localhost:12000', 'Accept': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.87 Safari/537.36', 'Connection': 'keep-alive', 'Upgrade-Insecure-Requests': '1'})
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/home/wallaceb/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/__main__.py", line 200, in get
    self.render('station_list.html', stations=self.store.stations)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 699, in render
    html = self.render_string(template_name, **kwargs)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 803, in render_string
    t = loader.load(template_name)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/template.py", line 424, in load
    self.templates[name] = self._create_template(name)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/template.py", line 451, in _create_template
    with open(path, "rb") as f:
IOError: [Errno 2] No such file or directory: '/home/wallaceb/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/templates/station_list.html'
ERROR:tornado.access:500 GET / (127.0.0.1) 5.35ms
WARNING:tornado.access:404 GET /favicon.ico (127.0.0.1) 2.58ms
ERROR:tornado.application:Uncaught exception GET / (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='GET', uri='/', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, sdch', 'Host': 'localhost:12000', 'Accept': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.87 Safari/537.36', 'Connection': 'keep-alive', 'Cache-Control': 'max-age=0', 'Upgrade-Insecure-Requests': '1'})
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/home/wallaceb/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/__main__.py", line 200, in get
    self.render('station_list.html', stations=self.store.stations)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 699, in render
    html = self.render_string(template_name, **kwargs)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 803, in render_string
    t = loader.load(template_name)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/template.py", line 424, in load
    self.templates[name] = self._create_template(name)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/template.py", line 451, in _create_template
    with open(path, "rb") as f:
IOError: [Errno 2] No such file or directory: '/home/wallaceb/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/templates/station_list.html'
ERROR:tornado.access:500 GET / (127.0.0.1) 0.70ms

jettisonjoe, 2016-04-05:
The new frontend will be built differently, so while I'm working on that I'll try to head this off.

wallacbe, 2016-07-03:
I don't see this from HEAD, closing

closed 2016-07-03 by wallacbe