google/openhtf #174: MFG inspector setting Status to "ERROR" regardless of Pass/Fail state

wallacbe, 2016-03-15:
CLI shows Pass/Fail result but MFG inspector is always setting "Status" field to "Error"

protoc --decode_raw < test.1458078080114.pb 
1: "test"
2: "wallacbe"
3 {
  1: "MyTest"
  2: "OpenHTF Example Test"
  5: "1.0.0"
}
**4: 3
**

grybmadsci, 2016-03-15:
I've made a few changes to `mfg_inspector.py` in #168 , can you try again when that's merged (doing it now, should be in any second).

wallacbe, 2016-03-15:
Cloned from head a minute ago, still see issue.

wallacbe, 2016-03-16:
pulled from head again.  Now seeing a different error when trying to output protobuf:

~/openhtf/examples$ python hello_world.py 
Provide a DUT ID in order to start the test.
test
Hello World!
Finishing test execution with outcome PASS.
2016-03-15 21:55:32,814 - ERROR - Failed to execute test.
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 136, in _RunTestLoop
    self.test.OutputTestRecord(self._test_state.GetFinishedRecord())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 79, in OutputTestRecord
    output_cb(record)
  File "/home/wallaceb/openhtf/examples/output.py", line 32, in OutputTestRun
    return output(record)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 294, in __call__
    outfile.write(_TestRunFromTestRecord(test_record).SerializeToString())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 261, in _TestRunFromTestRecord
    _PopulateHeader(record, testrun)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 58, in _PopulateHeader
    testrun.tester_name = record.station_id
  File "build/bdist.linux-x86_64/egg/google/protobuf/internal/python_message.py", line 471, in field_setter
    self._fields[field] = type_checker.CheckValue(new_value)
  File "build/bdist.linux-x86_64/egg/google/protobuf/internal/type_checkers.py", line 155, in CheckValue
    raise TypeError(message)
TypeError: None has type <type 'NoneType'>, but expected one of: (<type 'str'>, <type 'unicode'>)
ERROR:root:Thread raised an exception: TestExecutorThread
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 109, in _ThreadProc
    self._RunTestLoop.once(self)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 136, in _RunTestLoop
    self.test.OutputTestRecord(self._test_state.GetFinishedRecord())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 79, in OutputTestRecord
    output_cb(record)
  File "/home/wallaceb/openhtf/examples/output.py", line 32, in OutputTestRun
    return output(record)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 294, in __call__
    outfile.write(_TestRunFromTestRecord(test_record).SerializeToString())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 261, in _TestRunFromTestRecord
    _PopulateHeader(record, testrun)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 58, in _PopulateHeader
    testrun.tester_name = record.station_id
  File "build/bdist.linux-x86_64/egg/google/protobuf/internal/python_message.py", line 471, in field_setter
    self._fields[field] = type_checker.CheckValue(new_value)
  File "build/bdist.linux-x86_64/egg/google/protobuf/internal/type_checkers.py", line 155, in CheckValue
    raise TypeError(message)
TypeError: None has type <type 'NoneType'>, but expected one of: (<type 'str'>, <type 'unicode'>)
Exception in thread TestExecutorThread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/threads.py", line 51, in run
    self._ThreadProc()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 109, in _ThreadProc
    self._RunTestLoop.once(self)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 136, in _RunTestLoop
    self.test.OutputTestRecord(self._test_state.GetFinishedRecord())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 79, in OutputTestRecord
    output_cb(record)
  File "/home/wallaceb/openhtf/examples/output.py", line 32, in OutputTestRun
    return output(record)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 294, in __call__
    outfile.write(_TestRunFromTestRecord(test_record).SerializeToString())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 261, in _TestRunFromTestRecord
    _PopulateHeader(record, testrun)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/output/mfg_inspector.py", line 58, in _PopulateHeader
    testrun.tester_name = record.station_id
  File "build/bdist.linux-x86_64/egg/google/protobuf/internal/python_message.py", line 471, in field_setter
    self._fields[field] = type_checker.CheckValue(new_value)
  File "build/bdist.linux-x86_64/egg/google/protobuf/internal/type_checkers.py", line 155, in CheckValue
    raise TypeError(message)
TypeError: None has type <type 'NoneType'>, but expected one of: (<type 'str'>, <type 'unicode'>)

grybmadsci, 2016-03-16:
Ah, `hello_world.py` was written with the new conf in mind, so it doesn't set a station_id, and there was no default.  Added the fallback to hostname, once #171 is merged this will still be fixed (same fix is in that branch).

grybmadsci, 2016-03-16:
Should be fixed.

closed 2016-03-16 by grybmadsci