google/openhtf #176: Measurement InRange() and MatchesRegex() Fail Values showing up as outcome PASS

wallacbe, 2016-03-21:
I'm setting measurements to a fail result and it's showing up as outcome PASS.

Code addition to hello_world.py:

`@measures(Measurement('hello_world_measurement').MatchesRegex(r'Bye'))`
`@measures(Measurement('foo').InRange(0,1))`
`def hello_world(test):`
`  """A hello world test phase."""`
`  test.measurements.hello_world_measurement = 'Hello Again!'`
`  test.measurements.foo = -1`

output:

python hello_world.py 
Provide a DUT ID in order to start the test.
test
Finishing test execution with outcome PASS.

Also should make adding unit test for this this case since pass/fail is critical item for testapp.

grybmadsci, 2016-03-21:
Did you verify in the JSON output the result is actually PASS?  That printed message may be printed before the measurement verification happens.  It definitely still needs a fix, but it would help track down what's going wrong.

wallacbe, 2016-03-21:
JSON shows fail for the measurements but pass for the outcome

"measurements": {
                "foo": {
                    "outcome": "FAIL",
                    "name": "foo",
                    "validators": [
                        "0 <= x <= 1"
                    ]
                },
                "hello_world_measurement": {
                    "outcome": "FAIL",
                    "name": "hello_world_measurement",
                    "validators": [
                        "'x' matches /Bye/"
                    ]
                }
            },

    },
    "log_records": [
        {
            "timestamp_millis": "1458529183819",
            "level": "10",
            "source": "test_state.py",
            "lineno": "146",
            "message": "Finishing test execution with outcome PASS.",
            "logger_name": "openhtf.test_record"
        }
    ],
    "outcome": "PASS",
    "metadata": {}

grybmadsci, 2016-03-21:
Ah, I remember seeing this and noting that I had to update it, but forgot to do so, should have put a TODO there.. You're right that we need unit tests that test fail and error cases, I think I said the same thing in a comment a while back - I'll create an issue to track that.

grybmadsci, 2016-03-21:
Fix merged!

closed 2016-03-21 by grybmadsci