google/openhtf #182: Add phase unit test example with new conf

wallacbe, 2016-03-22:
Old conf unit test method doesn't work:

```
import unittest
import mock
import hello_world
import openhtf

class TestPhases(unittest.TestCase):

  @mock.patch('openhtf.conf.Config', autospec=True)
  @mock.patch.object(openhtf.exe, 'phase_data')
  def test_hello_world(self, mock_test, mock_config):
    hello_world.hello_world(mock_test)
    mock_config.return_value.foo = 'bar'

if __name__ == '__main__':
  unittest.main()
```

Removed the config mock decorator and used Load() instead.  Does this seem like a reasonable way  to mock the config?

```
import unittest
import mock
import hello_world
import openhtf

class TestPhases(unittest.TestCase):

  @mock.patch.object(openhtf.exe, 'phase_data')
  def test_hello_world(self, mock_test):
    openhtf.conf.Load(foo='bar')
    hello_world.hello_world(mock_test)

if __name__ == '__main__':
  unittest.main()

```

fahhem, 2016-03-22:
Unfortunately, as is, that code is affecting other tests because you're not resetting the config and other tests may accidentally depend on the order.

You can add a `self.addCleanup()` call:

```py
    openhtf.conf.Load(foo='bar')
    self.addCleanup(openhtf.conf.Reset)
```

That will reset the state correctly

fahhem, 2016-11-25:
The way tests work has been completely overhauled by @grybmadsci, closing this issue since there are now several examples of unit-testing (such as using `conf.save_and_restore`)

closed 2016-11-25 by fahhem