google/openhtf #188: Audit pylintrc file

grybmadsci, 2016-03-23:
Let's do a once-over of our pylintrc file and then we can set a little time aside for pylinting the whole codebase.

maddychan, 2016-06-28:
Note: The current pylintrc file has the indentation set to two whitespaces, which is the internal standard but not the external standard (4 spaces). We're holding off on updating that right now because it would mean a lot of restructuring code, but it's something to keep in mind

grybmadsci, 2016-06-28:
Yeah, I noticed that the other day... I personally prefer 2-space indent.. 80 characters is already somewhat restrictive once you start using more descriptive names for things :(  But, we really should stick with the standard (I believe externally Google has standardized on 4 spaces, even though there's a ton of legacy code internally on 2 space indents).

Refactoring that would be one way to bump your line contribution count ;)  There are certainly higher priorities though...

jettisonjoe, 2016-06-28:
4 spaces _pains_ me :-( but yeah, we should probably update eventually. :cactus:

grybmadsci, 2016-06-28:
LOL :cactus: needs to be an option in "add your reaction"...

jettisonjoe, 2016-07-19:
Done. Thanks, @maddychan!

closed 2016-07-19 by jettisonjoe