google/openhtf #189: flag a measurement as FAIL if the returned value is 'nan'

song7seas, 2016-03-24:
Can we flag a measurement as FAIL if the returned value is ‘nan’, if the measurement is assigned the pass/fail limits? Currently it’s marked as “PASS” as the example below. 

"measured_values":{"foo":"nan"},

"foo":{"units":{"uom_suffix":"dBmW","uom_code":"2N"},"outcome":"PASS","docstring":"blah","name":"foo","validators":["0 <= x <= 10"]},

grybmadsci, 2016-04-04:
That's definitely a bug, someone will take a look when we get some time :)

wallacbe, 2016-05-22:
Any thoughts on handling -inf and inf?

It will behave like a number within InRange and fail if InRange(1, 4).  However the case where InRange(1, None) inf will pass.  However inf is sometimes an indicator something went wrong, (like in FFTs).  Should we reject inf or treat it like a normal float?

grybmadsci, 2016-05-22:
I think inf and -inf should pass, it's up to the author to specify an upper
limit if there is one, perhaps using something like sys.maxint if you want
"really large but not infinite upper bound".

fahhem, 2016-05-22:
sys.float_info.max is the true max, so I think it's reasonable if we provide a shortcut:

`RealNumber = InRange(-sys.float_info.max, sys.float_info.max)`

jettisonjoe, 2016-05-24:
Maybe fixed by #264. Separate feature request for a `RealNumber` validator?

wallacbe, 2016-05-24:
Moved RealNumber/inf conversation into:
https://github.com/google/openhtf/issues/266

I verified that https://github.com/google/openhtf/pull/264 fixes this issue. should close

closed 2016-05-24 by jettisonjoe