google/openhtf #190: Add logger to plugs

wallacbe, 2016-03-24:
In phase we can call logger like: test.logger.info('foo')

In plugs we have to import logging and call it like logging.info()

can a logger be added to plugs.BasePlug class so we don't have to import?  Also observing that logging.info('foo') doesn't output to the terminal when called in a plug

grybmadsci, 2016-03-24:
Hm, yeah I have encountered this myself, (wanted better logging integration inside plugs), and it's kind of frustrating.  You can manually do `logging.getLogger(logs.RECORD_LOGGER)` to get the logger that will output to the `test_record`, but it's annoying to have to, it's a good idea to add it to `BasePlug` as a `self.logger` attribute.

As for the not-outputting to a terminal, I'm not sure what that is, but I'm seeing some weird behavior with terminal output too, in #142 , and we don't have a ton of info on the problem, so if you have specific reproducible examples, could you add them there?

wallacbe, 2016-03-24:
If I use your recommendation I see it in the terminal.
   logger = logging.getLogger(logs.RECORD_LOGGER)
   logger.info('Tear down')

Yeah it would be nice to have it in the baseplug.  There may be cases where someone wants it displayed other times not.

jettisonjoe, 2016-03-24:
I thought our intention was to have plug logs be sent to a log file along with framework logs (rather than to the test record), although I do admit plugs are in a grey area between framework and test code. Regardless, it would be cool to hook up one or both loggers in `BasePlug`.

grybmadsci, 2016-03-24:
That was our intention as a default behavior, but I could see users wanting to log to either the framework logs ('openhtf' prefix) or the test record directly, so I think it would be neat to provide them as attributes on the BasePlug.

I am a little concerned that we're doing something somewhere that's messing with the root logger, because you shouldn't *have* to use one of the openhtf loggers to see your logs...

jettisonjoe, 2016-03-24:
We can probably find ways to do everything we need to do without messing with the root logger. ...Right?

grybmadsci, 2016-03-24:
Yeah, I believe we currently aren't (intentionally) touching the root logger, but brandon and I are both seeing some weird behavior, so we need to look into it.  Ideally, yeah we don't touch the root logger at all.

jettisonjoe, 2016-03-24:
Ah, I see. Agreed.

Back on point, I'm onboard with providing access to one or both loggers as attributes of `BasePlug`. @wallacbe, that was a really good idea, actually.

wallacbe, 2016-05-21:
Add __init__ to BasePlug and super to plugs' __init__
```
  def __init__(self):
    self.logger = logging.getLogger(logs.RECORD_LOGGER)

  @conf.InjectPositionalArgs
  def __init__(self, example_plug_increment):
    super(ExamplePlug, self).__init__()
    self.increment = example_plug_increment
    self.value = 0

```

closed 2016-05-22 by wallacbe