google/openhtf #202: Enforce measurement name sanity.

grybmadsci, 2016-03-29:
measurement names should be valid Python variable names.

can't assign @fahhem for some reason, but he's on it.

fahhem, 2016-03-29:
I'm guessing you can't be assigned if you're not 'on' the project.

grybmadsci, 2016-03-29:
Weird I thought you were listed as a collaborator already, whoops!  Fixed and assigned :)

fahhem, 2016-04-21:
I'm not sure I agree with this after all. I can see the appeal, but is there a technical reason to force this? I think it can make sense to have measurements named "Current Draw" over "current_draw" if your team wants it. And now that we allow both setting and getting via dictionary access, we don't even have that technical reason.

jettisonjoe, 2016-04-25:
I personally _like_ the enforcement of python names for measurements, but it's mostly aesthetic (and possibly just general wisdom) rather than from a specific technical reason. It bothers me that if you have measurements named `Current Draw` and `Voltage`, for example, that `Voltage` will be accessible via the `.` syntax but `Current Draw` won't. We should at the very least consider logging a warning if a measurement will break `.` syntax access.

grybmadsci, 2016-04-25:
Hm, I think the Doc() mechanism provides a means to include any more-human-readable text than the python variable name provides.  My problem with allowing arbitrary strings is that it's a slippery slope; measurements are the equivalent of variables in OpenHTF land, and by allowing arbitrary strings, you're setting people up for doing *really* stupid things.

I've actually had places where I've seen people track all their variables in a dict and use strings to access them:
```
my_vars = {
 'My First Var': 4,
 'I am a Window Programmer': 10,
}
...
def foo():
  my_vars['Because Why Not'] = my_vars['My First Var'] + my_vars['fml what is keyerror']
```
There's no technical reason why that's incorrect - it works perfectly fine, but it's obviously wrong.

I view allowing arbitrary strings as measurements to be roughly equivalent.  The only reason dict access is supported at all is to avoid having to use getattr when measurement names are programmatically generated, similar to something like `locals()['my_weird_%s_name' % weird]`, but a little less awkward and more common.

closed 2022-10-13 by glados-verma