google/openhtf #203: consider passing entire test_record to test_stop

grybmadsci, 2016-03-29:
right now `test_stop` triggers get passed the dut_id only, but there may be other information in the `test_record` that the stop trigger may want to use.  In my case, I got around this by instantiating a class that provides both an output callback and a `test_stop` trigger and stashing the test_record when the former is called for accessing in the latter.

i'd like to propose we update the calling convention of `test_stop` triggers to take the entire `test_record` (note that because it's already been output, this is a read-only view of the `test_record`).  Thoughts?  Does this break anyone?

jettisonjoe, 2016-03-29:
I don't think it will break much if anything, but it might become a source of confusion about where finished test records "with a bow on them" come from. I can see people mistakenly it as a sort of output module.

fahhem, 2016-03-30:
I can see wanting certain data from the test_record, like whether it passed or
how long it took. Was there something in particular you needed? I'd like to come
up with a way to avoid Joe's situation where there's an unfinished test_record
in test_stop and a finished one in OutputCallback and they may use the wrong
one?

grybmadsci, 2016-03-30:
Basically, I pull out all sorts of things to display an 'end of test summary' GUI (i've been forced to provide GUI interfaces /cry).  By the time test_stop gets called, output callbacks have already been called and are done with.  Ideally, we'd provide a read-only view of the test_record to test_stop, but there's no easy way to do that, unless you want to add a readonly mechanism to mutablerecords ;)

I suppose *really* I should be creating that GUI as an output callback, and my test_stop should block on that GUI being closed, rather than lumping those two things into one.  I'm not sure dut_id is really useful to pass into test_stop, though.  Why is that more useful than anything else?  If you're waiting for a USB device to go away, for example, then you probably want access to a plug?

I think we need to rethink test_stop a tad...

fahhem, 2016-03-30:
test_stop taking a DUT id seems to be a very USB-specific thing. The only thing
I've put in there is a sleep for looping, I usually put cleanup stuff in
TearDown of a plug or as one of the last phases.
A diagram of what happens where would probably be a great start to the docs :)

jettisonjoe, 2016-03-30:
Just throwing out wild ideas here; what about blowing away stop triggers and putting in the ability to mark some output modules as blocking? Or having a generic `OnDone` that gets passed all the same things phases get passed?

grybmadsci, 2016-03-30:
Test stop triggers are poorly named, it's actually a test re-start trigger.  The test is output and done with before we wait on the trigger, the triggers is to prevent auto-looping right away and retesting the same device infinitely, and is a holdover from XTF.

I would like to remove test looping from OpenHTF entirely, really.  It's a scheduling decision (what to test when), and is outside the scope of OpenHTF.  If that happens, test_stop can just go away.  I would be totally fine with that approach.

This would force users who *want* to loop to do a little more boilerplate, so instead of just calling `Test.Execute()`, you have:
```
while True:
  Test.Execute()
  MyQuoteStopTriggerUnquote()
```

I think that's probably fine and is much clearer than `loop=True` combined with a poorly named `test_stop` callback.

grybmadsci, 2016-03-30:
In fact, a neat side project might be to implement a rudimentary scheduler to ship as a utility with openhtf in an extra or somesuch.

jettisonjoe, 2016-03-30:
```
#!/bin/bash

crontab -e
```

Nailed it.

fahhem, 2016-03-30:
The over-engineered, fewer features version of cron. You know, what self-respecting engineer would do to show impact 😆

```py
while True:
  test.Execute()
  time.sleep(60*minutes) 
```

grybmadsci, 2016-04-04:
All right, edited to get to the root of the issue here.  Two main questions to answer then I'll make this a feature request or close it out:
 - Is anyone currently using looping that isn't comfortable with how to handle it going away (@wallacebe have some thoughts here?)
 - Do people think looping is fundamentally part of OpenHTF's stated goal, and therefore should stay.

fahhem, 2016-04-04:
If there was some feature looping provides that we can't easily expose otherwise, I would be in favor of it. There is one downside I foresee, which is that they will incur the HTTP server shutdown/startup on each cycle. However, I'm working on making that much quicker, so I don't know how much folks care beyond that.

wallacbe, 2016-04-05:
I did a quick check with a while loop in py_binary, appears to have similar looping behavior.
```
while True:
    tester.Execute(test_start=triggers.PromptForTestStart())
```

If this is removed I would recommend adding a looping example to the examples.

grybmadsci, 2016-04-05:
Ah right, I was also considering pulling the http_api stuff out of Execute and putting it in Initialize instead (and making it disable-able via argument to Initialize), that way if you run multiple Tests, either in a loop or in parallel, you don't get multiple http_api start/stops.

Currently, the http_api is forced-singleton because it opens a port that's set via flags (also probably should pull that out of flags and pass it in).  I think it's fine to keep it singleton if we make it play a little nicer with multiple Test()'s running at a time.  This is one way to do 'multi-cell', for example.  Not sure it'll get used, but i think it's a cleaner design that way regardless.

Thoughts?

jettisonjoe, 2016-04-05:
Since the HTTP API is sort of a framework-level construct, I think it makes sense to put it in `Initialize` instead, although it will take a little bit of work to support multiple tests with a singleton HTTP API.

grybmadsci, 2016-04-27:
Done!

closed 2016-04-27 by grybmadsci