google/openhtf #204: Examine plug instantiation rules

grybmadsci, 2016-03-29:
Currently, plugs get instantiated each time they are used with a new key.  For example:
```
@plug(foo=MyPlug)
def PhaseOne(test, foo):
  pass

@plug(bar=MyPlug)
def PhaseTwo(test, bar):
  pass
```
Will result in *two* instances of MyPlug being created, but:
```
@plug(foo=MyPlug)
def PhaseOne(test, foo):
  pass

@plug(foo=MyPlug)
def PhaseTwo(test, foo):
  pass
```
Results in only one instance being created.  Notably:
```
@plug(foo=MyPlug)
def PhaseOne(test, foo):
  pass

@plug(foo=MyOtherPlug)
def PhaseTwo(test, foo):
  pass
```
Results in an error indicating that you can't have two different types for the same plug name, even in different phases.

The single-instance behavior is necessary because most plugs are pseudo-singletons (they aren't *actually* singletons, but if they open up any hardware ports, they likely will fail to instantiate if you try to make multiple).

The question I would like to pose is, do we want to allow the creation of a second instance of a plug by assigning it to a different name?  It seems an odd loophole in the otherwise-singleton model we have for plugs.  In fact, the recommended way of getting multiple of the same plug (presumably with some config change) is to create subclasses of the parent plug, so that there is *still* only one instance of each of those subclasses.

I'd like to suggest that we either:
 -  Error out if you try to use a different name for the same plug type.
 -  Make that second name an alias to the shared instance, and stay consistent with singleton-like behavior.

Thoughts?

grybmadsci, 2016-03-29:
As a side note, we could also reconsider the error generated by that last code block.  If we index plug instances by their respective types and only give that key name meaning within the context of the single phase, then you can use whatever name you want for whatever plug type you want for any given phase, and we handle the singleton-ification of plugs at the PlugManager level.

jettisonjoe, 2016-03-29:
The consensus based on this morning's meeting was to make plug instances (at least ones instantiated by the framework) effectively singletons for the duration of a test run. Decorating for the same plug class with a different name should still result in the same instance getting passed in. A `MyPlug` by any other `foo`, etc.

fahhem, 2016-03-30:
I like this, so the plug manager would just take a list of plug types and the
phase executor would say “I need these plugs now” and do the mapping itself?

grybmadsci, 2016-03-30:
Exactly, the plug manager keeps them by type, and the phase executor actually does the arg name to plug instance mapping.  Sounds like we're all in agreement, so I'll hop on this sometime soonish :)  It's EOQ right now, so time may be limited until next week.

grybmadsci, 2016-04-04:
Okay looks like we're all on the same page for this, will try to get to it this week sometime.

closed 2016-04-27 by grybmadsci