google/openhtf #210: consider adding docstrings to measurement dimensions

grybmadsci, 2016-04-04:
not sure what the syntax around this would look like just yet, but i'm finding myself looking at my own test results and realizing if i didn't already know what the dimensions were, the units alone wouldn't be quite enough description to understand what i'm looking at.

grybmadsci, 2016-04-04:
any thoughts?  is this something people would use?  what syntax might it use?

jettisonjoe, 2016-04-05:
I think measurement dimensions are kinda hard to understand a lot of the time, especially if you're not the original author of the test (and probably even if you are). Docstrings might help that. My holdback is that the measurement declaration syntax is already pretty verbose and easy to get lost in itself.

closed 2022-10-13 by glados-verma