google/openhtf #219: Phase unit tests - mock plugs and test for outcome

wallacbe, 2016-04-09:
Documenting the code for mocking plugs and testing for outcome.
```
def ReplaceType(phase, old_type, new_type):
  phase = openhtf.PhaseInfo.WrapOrCopy(phase)
  for plug in phase.plugs:
    if plug.cls == old_type:
      plug.cls = new_type
  return phase

def _RunPhase(phase, plug, mock_plug):
  result = util.NonLocalResult()
  phase = ReplaceType(phase, plug, mock_plug)
  test = openhtf.Test(phase)
  def _SaveResult(test_record):
    result.result = test_record
  test.AddOutputCallback(_SaveResult)
  test.Execute()
  return result.result

 def testPhase(self):
    mock_plug = mock.Mock()
    test_record = _RunPhase(PhaseFoo, SomePlug, mock_plug)
    self.assertEqual(test_record.outcome.value, measurements.Outcome.FAIL.value)

```

wallacbe, 2016-04-15:
Had to add the following function at the end of my test cases:

def _ResetPhase(phase, plug, mock_plug):
  _ReplaceType(phase, mock_plug, plug)

wallacbe, 2016-04-18:
It would be preferred to have a way to look at both outcome and measurements using the same procedure.  Currently have to perform unit tests differently to do that.  (this procedure vs calling phase directly)

wallacbe, 2016-05-16:
One other request is to be able to mock the serial number in the test object

wallacbe, 2016-05-17:
Also should disable http_port by default

test.Configure(http_port=None)
test.Execute(test_start=lambda: 'TestDUT')

jettisonjoe, 2016-05-19:
I think @grybmadsci was gonna look at the unit testing stuff. @grybmadsci, please feel free to unassign if I'm wrong.

wallacbe, 2016-05-22:
Here's an idea.  We can add a TestCase subclass to utils.  Exception, outcome and measured_values would be accessible as class attributes.  Outcome constants would be available to compare againts in assertions.  Would like some input on how to handle prompts since those hold up unit test execution.

class TestCase(unittest.TestCase):

  PASS = test_record.Outcome.PASS.value
  FAIL = test_record.Outcome.FAIL.value
  ERROR = test_record.Outcome.ERROR.value

  def _ReplaceType(self, phase, old_types, new_types):
    phase = openhtf.PhaseInfo.WrapOrCopy(phase)
    for plug in phase.plugs:
      for old_type, new_type in zip(old_types, new_types):
        if plug.cls == old_type:
          plug.cls = new_type
    return phase

  def RunPhase(self, phase, plugs=[], mock_plugs=[], serial='test'):
    result = util.NonLocalResult()
    phase = self._ReplaceType(phase, plugs, mock_plugs)
    test = openhtf.Test(phase)
    def _SaveResult(tr):
      result.result = tr
    test.AddOutputCallbacks(_SaveResult)
    test.Configure(http_port=None)
    test.Execute(test_start=lambda: serial)
    self._ReplaceType(phase, mock_plugs, plugs)
    if result.result.outcome_details:
      self.exception = result.result.outcome_details[0].code
    else:
      self.exception = None
    self.outcome = result.result.outcome.value
    self.values = result.result.phases[0].measured_values

closed 2016-09-20 by wallacbe