google/openhtf #220: Logs are appending from previous test run

wallacbe, 2016-04-10:
Added a loop to example tester, I'm seeing logs append:

<class 'example_plug.ExamplePlug'>
Finishing test execution with outcome PASS.
<class 'example_plug.ExamplePlug'>
Finishing test execution with outcome PASS.
Finishing test execution with outcome PASS.
<class 'example_plug.ExamplePlug'>
Finishing test execution with outcome PASS.
Finishing test execution with outcome PASS.
Finishing test execution with outcome PASS.

grybmadsci, 2016-04-10:
Hm, this was an old issue that I thought we'd fixed.. Can you verify that you've got the latest from the master branch?

wallacbe, 2016-04-10:
Did a fresh clone of the repo and confirmed I see the same issue.

I'm trying to pull the looping mechanism out of the execute() in anticipation of that feature being removed.

In hello_world main()  I added a loop:

```
if __name__ == '__main__':
  for i in range(0, 3):
    test = Test(hello_world)
    test.AddOutputCallback(
       json_factory.OutputToJSON('./hello_world.json', indent=4))
    test.Execute()
```

fahhem, 2016-04-10:
Ohh, we're not ready for that yet, SetupFramework() gets called for every
Execute() call, so that's what's appending the loggers (the call to
logs.setup_logger()).
I guess we could quickly fix that by adding @functions.RunOnce() onto
SetupFramework(), then this won't happen

grybmadsci, 2016-04-10:
Ah, I see what happened.  Good call on getting ahead of the game, @wallacbe and thanks for the quick fix @fahhem :D Will merge that as soon as tests pass.

closed 2016-04-10 by grybmadsci