google/openhtf #222: Network is unreachable error when network is disconnected

agdeepak, 2016-04-12:
Local Front-end throws network is unreachable error when there is no network connected or the network drops while running front end. This causes front end to crash and it needs to be restarted again or wait until network is connected. 

Issue can be easily reproduced by disconnecting any wifi or ethernet network and then starting front-end locally.
Please see error screenshot attached.
[image]

jettisonjoe, 2016-04-13:
I'll look into fixing this.

jettisonjoe, 2018-05-04:
Closing this as obsolete because there's a whole new different frontend now!

closed 2018-05-04 by jettisonjoe