google/openhtf #223: Front-end IP conflict when 2 LAN networks are connected

agdeepak, 2016-04-12:
(Sighting)
Use case (steps to reproduce):
Connect lab ethernet network on LAN port
Connect another device like USB hub using USB via ethernet adapter
Start Front-end. Error shows IP conflicts as 2 different IPs are assigned for each connected network.

Impact: Unable to run front end when ethernet and USB hub via ethernet connected simultaneously.
Will update the bug with the screenshot during next testing phase.

jettisonjoe, 2016-04-13:
Screenshot or console copy/paste would be helpful.

jettisonjoe, 2016-06-06:
I have two network interfaces I can set up on my machine. I will attempt to repro.

grybmadsci, 2016-06-06:
I suspect this requires the ability to bind any ports we listen on to a specific interface (via `conf` seems reasonable).  I've already built this into the new up-and-coming `station_api`, you'll need to do something similar in the frontend backend (lol) @jettisonjoe .

jettisonjoe, 2016-06-07:
We might want to check on multicast, too, although I think I wrote that to explicitly listen on _all_ interfaces.

jettisonjoe, 2016-07-21:
Confirmed tornado-based frontend now listens on all interfaces by default. Starting the frontend with two networks connected no longer produces an error.

closed 2016-07-21 by jettisonjoe